tzapu/WiFiManager #85: can't connect after config

yomasa, 2016-01-29:
can't connect after config so never get  to save added parameters.
but connects fine after hard reset.

here is the output

*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: server
*WM: m10.cloudmqtt.com

yomasa, 2016-01-31:
after more testing, I found that it only fails to connect after config when using the wifiManager.startConfigPortal, I using a button to start wifiManager.startConfigPortal

Thanks
Jorge

tzapu, 2016-02-04:
hi,

i think you can twist the example a bit so you always save after it returns if connection is unsucessful, right before restart.

will that work?

probonopd, 2016-02-07:
How is this supposed to work? Currently with [this sketch](https://github.com/probonopd/sketches/blob/master/esp8266-neopixelsanimator-ota-http/esp8266-neopixelsanimator-ota-http.ino) "almost" everything works, but after I have entered the WLAN password and pressed "save" I seemingly need to power cycle the device for the sketch to become responsive to HTTP requests.

tzapu, 2016-02-08:
there s some weirdness with some modules, they just won t connect and i don t know why...
but in your case, it s probably more of the mix of your own http server, which other users have reported not working together with wifimanager, need to get a test setup...

tzapu, 2016-02-08:
btw @probonopd for casese like this i added an example AutoConnectWithReset, that resets it automatically after config, if it did not connect, although i assumes yours connects and needs the reset just so the http server starts working...

probonopd, 2016-02-08:
I get
```
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Ready
IP address: 192.168.0.19

Running...
HTTP server started
```
But then the HTTP server is not reachable until I reboot.

Here is my setup():

```
void setup()
{

  Serial.begin(115200);
  Serial.println("Booting");
  
  // WiFiManager AutoConnectWithTimeout
  WiFiManager wifiManager;
  wifiManager.setBreakAfterConfig(true); // exit after config instead of connecting
  wifiManager.setTimeout(180); //  timeout until configuration portal gets turned off
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    ESP.reset(); // reset and try again, or maybe put it to deep sleep
    delay(5000);
  } 
  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.println();
  Serial.println("Running...");

  httpserver.onNotFound(handleRequest);

  httpserver.begin();
  Serial.println("HTTP server started");

  // Advertise http service
  MDNS.addService("http", "tcp", 80);
}
```

Am I doing something wrong? Adding `wifiManager.setBreakAfterConfig(true);` did not seem to make a difference at all. Is there a way to reboot the device then the user has entered the credentials and they are saved?

tzapu, 2016-02-09:
wifiManager.setBreakAfterConfig(true); will just make wifimanager exit after it tries to connect even if unsuccessful. In your case, you would need to add ESP.reset() after it exits, no matter the outcome, you can set a flag in the start config portal or save configuration callbacks (so you know it actually went through a configuration phase, so you only reset then).
Credentials should be saved up to that point.
In your code, it will probably connect, then exit and never get to Serial.println("failed to connect and hit timeout");. 

btw, i did get a working version of the web server like this https://gist.github.com/tzapu/ecc0759829d30d5a6152

igrr is also looking into it to see why the webserver doens t start a second time.

let me know if the above helps
cheers

eppi72, 2016-02-15:
hi tzapu
is it possible that you can create a sample code for web OTA? I would like a update Button for upload sketch (compiled bin file) with a Button "update" under the other buttons (Wifi Config, Wifi Config "no scan", Reset, Info). Thank you verry much - Cheers

probonopd, 2016-02-15:
> In your case, you would need to add ESP.reset() after it exits, no matter the outcome

Thanks @tzapu - how would I do this in the sketch without editing the lib itself?

tzapu, 2016-02-16:
@eppi72  will think about it, i m not geared for ota currently so need a better test bed
the other option is to provide customisable buttons

@probonopd 
use `wifiManager.setBreakAfterConfig(true); ` this will make it exit after a connection attempt, even if not successful
if the save callback was called go through the code that saves the params
then reset

closed 2018-01-26 by tablatronix