tzapu/WiFiManager #86: OnDemand Portal unresponsive when also running webserver

knnniggett, 2016-01-30:
I'm not convinced the cause of this issue is in the WiFiManager, but this environment seems to be the best place to get a better understanding of what is going on here.

The following test was done with an ESP-01.

Take the OnDemandConfigPortal example sketch and load it as-is into the ESP8266. Observe that it works as intended. Pushing a button wired to GPIO0, causes the OnDemand portal to start, and one can access the portal from a wireless client device and set the ssid parameters.

Next, modify the sketch to create and configure an ESP8266WebServer object. Then call `server.begin()` in setup().  Assuming the sketch also has the necessary logic added to attach to your wifi network and obtain an ip address, observe that one can successfully ping the ESP8266 and that the webserver responds when pointing a client browser to the device's ip on your local wifi network.

Now press the button wired to GPIO0. One can see the serial monitor report the OnDemand portal has successfully started, the monitor reports the ip address changed to 192.168.4.1, and one can connect to the ESP8266 Access Point from a wireless client device. The client device receives an ip address (usually 192.168.4.2), but the ESP8266 does not return a ping to 192.168.4.1 and naturally one cannot access Config Portal.

To remedy this, I modified the sketch to delete the former server object before starting the WiFiManager (tried both `delete &server;` and `server.~ESP8266WebServer();`. This made some progress as I can now ping the On Demand Portal, but the web portal is still unresponsive when I point my browser to 192.168.4.1. Additionally, the serial monitor does not indicate it is serving pages to a client like it normally does when things are working.

This is where I became totally befuddled. Deleting the server object should have fixed this. At least that is what I thought.

To continuing troubleshooting, I have tried several combinations  of the following, in different orders, before starting the WiFiManager:
```C
    server.~ESP8266WebServer();
    delete &server;
    WiFi.softAPdisconnect(true);
    delay(500);
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(500);
    WiFi.disconnect();
```
Other than deleting the server object, none of the other commands appeared to make any difference.

Note that if I comment out the call to `server.begin()` in my sketch, then the On Demand portal works correctly.  So logically it would seem that starting the initial web server is changing something external to the server object, which isn't getting reset when I attempt to start the OnDemand portal. I just don't know what that could be.

Incidentally, I can successfully accomplish the same thing using the AutoConnect portal. Everything works as intended, but of course in that scenario the AutoConnect portal is called before I start my web server so everything works as expected.

For reference, here is the sketch I am using:
https://gist.github.com/knnniggett/6268b575e05236372196

tzapu, 2016-01-31:
hi, it must surely be a conflict between the internal webserver and yours.

somebody told me that calling server.reset(); on the webserver should clean up after it, may wanna try that before delete..

knnniggett, 2016-01-31:
Unfortunately, the ESP8266WebServer class does not have a member called reset. Neither does the WiFi class. Was it some other class perhaps?

tzapu, 2016-02-01:
hi,

it seems to be more a feature of smart pointers
http://en.cppreference.com/w/cpp/memory/unique_ptr

see how web server is defined in WiFiManager, it uses reset to allocate and deallocate.

As soon as I get some time for this, I will try to build a webserver example myself…

hope you can make something of the above, memory management really does my head in…

knnniggett, 2016-02-01:
Ah, thanks for the link. I will read up on that. 

Previously, I did notice this in in WiFiManage::startConfigPortal `server.reset(new ESP8266WebServer(80));`. I'll model what I do against that example.

tzapu, 2016-02-08:
hi, may be of use to you, this is a way in which i could get webserver running after auto connect
https://gist.github.com/tzapu/ecc0759829d30d5a6152

closed 2016-02-16 by tzapu