tzapu/WiFiManager #87: Keep trying to connect

x10iman, 2016-01-30:
Thank you for this great lib which is very useful.My issue is that If for any reason there is power failure and both router and ESP goes down then if the ESP powers on before the router it enters the configuration mode and must be connected manually through the portal.
I have modified the WiFiManager.cpp along with OnDemandConfigPortal example to avoid such scenario.
Then ESP will enter a loop trying to connect to router until it gets connected. Then i moved the trigger pin function from the main loop to Setup function to call the portal in case the router WiFi credentials have been changed and then we can connect with new credentials or new network by holding trigger pin during boot time (factory default mode). 
Please support if there is a better coding for my modifications.
Thanks again

WiFiManager.cpp :

`boolean WiFiManager::autoConnect(char const *apName, char const *apPassword) {
  DEBUG_WM(F(""));
  DEBUG_WM(F("AutoConnect"));

  // read eeprom for ssid and pass
  String ssid = getSSID();
  String pass = getPassword();

  // attempt to connect; should it fail, fall back to AP
  WiFi.mode(WIFI_STA);
  if (connectWifi(ssid, pass) == WL_CONNECTED)   {
    DEBUG_WM(F("IP Address:"));
    DEBUG_WM(WiFi.localIP());
    //connected
    return true;
  }

  //return startConfigPortal(apName, apPassword);

    return autoConnect(apName, apPassword);  //try to connect again automatically}`

  `
 WiFiManager::startConfigPortal(char const *apName, char const *apPassword) {
.
.
.
 // using user-provided  _ssid, _pass in place of system-stored ssid amd pass
      if (connectWifi(_ssid, _pass) != WL_CONNECTED) {
        DEBUG_WM(F("Failed to connect."));
		
		return autoConnect(apName, apPassword);	// try to connect again automatically 	
 `
OnDemandConfigPortal.ino : 

`
#define TRIGGER_PIN 4 // cant be GPIO 0 nor 2
void setup() {
  Serial.begin(115200);
  pinMode(TRIGGER_PIN, INPUT);

  WiFiManager wifiManager;

  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    if (!wifiManager.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.reset();
      delay(5000);
    }
  }
  wifiManager.autoConnect("AutoConnectAP");// When power up try to connect using saved //credinital 
  Serial.println("connected...yeey :)");
}`

tzapu, 2016-01-31:
hi, you can already do this by just calling WiFi.begin() in loop in your sketch, on top of the OnDemandPortal example, with none of these modifications...

x10iman, 2016-02-01:
i tried that but but when i connect to router it failed and this what i got from serial monitor:

Starting

Exception (29):
epc1=0x402138f7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

ctx: cont 
sp: 3fff0140 end: 3fff04f0 offset: 01a0

>>>stack>>>
3fff02e0:  3ffe9330 80af1999 3fff3078 3fff0998  
3fff02f0:  3fff0310 40213a07 3ffed6ec ffffff1a  
3fff0300:  40230f89 00000017 60000200 42d60000  
3fff0310:  00000018 3ffede0c 3ffe83a8 3ffe83a8  
.
.
.
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

tzapu, 2016-02-04:
btw, you could also use an auto connect, and setTimeout

i use this on my battery powered devices, if they come up when router is down, they wait 2 mins, if no connection, they go back to sleep, they could reset and try again in your case. they would never get stuck this way

cheers

tzapu, 2016-02-04:
closed as retry ability is mentioned here 
https://github.com/tzapu/WiFiManager/issues/87

closed 2016-02-04 by tzapu

x10iman, 2016-02-04:
Yes this way is very useful for saving power. But when the router is up again do I have to reset the ESP or it wil wake up automatically?
Sorry I am not that good in programming!

tzapu, 2016-02-05:
when the timeout occurs, wifi manager will return with false and you can do an ESP.reset(). this will restart the esp and retry connecting. instead of reset you can send it to sleep for an arbitrary period, and when it wakes up it tries again.