tzapu/WiFiManager #88: Start WebServer after config

k7vlad, 2016-01-31:
As far as I can see, you can't start a new WebServer immediately after the configuration is done (ESP connects to the desired network). Is there any other good way to do that, without using ESP.reset(), as it works after a reboot?

tzapu, 2016-02-01:
i ve had one other report of this, i will look into is as soon as i get some time
are you using autoConnect or startConfigPortal ?

k7vlad, 2016-02-01:
I use autoconnect. Would it be possible to use the same "server" which is declared in your library by making the pointer public? I haven't tried it yet

tzapu, 2016-02-01:
hi, no, the library deallocates that at the end to save memory

cheers

tzapu, 2016-02-08:
hi , give this a try
https://gist.github.com/tzapu/ecc0759829d30d5a6152

k7vlad, 2016-02-13:
Brilliant. Thanks (buna treaba!)

closed 2016-02-13 by k7vlad

tzapu, 2016-02-14:
mersi :)