tzapu/WiFiManager #90: portal suddenly no longer works (failing port 80 sessions)

flok99, 2016-01-31:
- I configured an esp to a certain ssid via your excellent wifi manager
- i then moved to a different physical location with different access points with other ssids
- there I connected to the esp over wifi. on serial I see that it connects succesfully. also my laptop says it has an ip address 192.168.4.2
- when i connect using either chrome or firefox (or even telnet to port 80!) on  192.168.4.1, it connects and then immediately disconnects without sending anything. not even an http header.
- in the serial connection I see:
nit ESP8266NeoPixel
Init NeoPixels
Init WiFi
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO STRT UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: Adding parameter
*WM: Aoife_server
Entered config mode
192.168.4.1
ESP8266NeoPixel
*WM: 
*WM: Configuring access point... 
*WM: ESP8266NeoPixel
add 1
aid 1
station: 0c:d2:92:3d:80:0b join, AID = 1
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
- as you see I added an explicit "reset settings" as earlier attempts also failed
- sometimes it logs an "handle root" message, but still a disconnect after a second or so wihout any data
- I tried this with 2 different esp8266 modules. same problem for both. both were configured initially at the other location

tzapu, 2016-02-01:
hi, what esp8266 core are you on?
what version of the library are you using?

most likely this is a bug that got that got fixed in both esp core and wifimanager latest versions ( staging and 0.7)

flok99, 2016-02-01:
Hi,

One like these:
http://nl.aliexpress.com/item/1pcs-ESP8266-remote-serial-Port-WIFI-wireless-module-through-walls-best/32280714838.html?isOrig=true#extend
ESP-01 I think it is.

I've upgraded the esp8266 plugin in arduino somewhere last week.
I also upgraded wifimanager 0.7 last night just before I send that bug
report.

tzapu, 2016-02-01:
hi, you might need to update again the esp8266 core to the latest github version or to staging rc2

see instructions on their github

cheers

tzapu, 2016-02-01:
i ve updated WiFiManger to 0.8. might sort some of your issues, try it if you can

flok99, 2016-02-01:
Hi,

The arduino library manager only shows version 0.7.
And for the esp I've got 2.1.0-rc2 installed.

flok99, 2016-02-01:
Ok I checked again and now 0.8 is available.
MUCH better! thanks a lot!
Only scanning for accesspoints takes minutes.
But apart from that; configuration well now works.

closed 2016-02-01 by flok99

flok99, 2016-02-06:
Hi,

I was too quick: I'm now at an other location and got the same problem
again.

Init NeoPixels
Init WiFi
*WM: Adding parameter
*WM: Aoife_server
*WM:
*WM: AutoConnect
bcn 0
del if1
usl
mode : sta(18:fe:34:d5:9e:57)
add if0
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
f r0, scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s                         <---
correct, it's not here
reconnect
f 0, scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
reconnect
f -180, scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
*WM: Connection result:
*WM: 1
del if0
usl
mode : softAP(1a:fe:34:d5:9e:57)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Entered config mode
192.168.4.1
ESP8266NeoPixel
*WM:
*WM: Configuring access point...
*WM: ESP8266NeoPixel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
chg_A1:-140
add 1
aid 1
station: 0c:d2:92:3d:80:0b join, AID = 1
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
chg_A1:-100
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
chg_A1:-60
*WM: WiFi save
*WM: Parameter
*WM: Aoife_server
*WM: 192.168.64.1
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster                         <-----
I've set a new ap but as you see below it will try to connect to the old
ssid ("vanheusdendotcom2")
mode : sta(18:fe:34:d5:9e:57) + softAP(1a:fe:34:d5:9e:57)
add if0
f r-60, LmacRxBlk:1
LmacRxBlk:1
scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
reconnect
f 0, LmacRxBlk:1
LmacRxBlk:1
scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
reconnect
f -180, LmacRxBlk:1
LmacRxBlk:1
scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
*WM: Connection result:
*WM: 1
*WM: Failed to connect.
reconnect
f r-60, scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
reconnect
f 0, scandone
no VANHEUSDENDOTCOM2 found, reconnect after 1s
reconnect

flok99, 2016-02-06:
Oh and the arduino ide does not speak about 0.9, only 0.8.

tzapu, 2016-02-07:
weird, maybe they did not update their db yet? a bit opaque on how you can see what they have received or not...

anyway, that seems ike the esp doesn t connect, wifimanager seems to be sending the info on properly...

tzapu, 2016-02-07:
is it a no password AP?

flok99, 2016-02-07:
That's correct.

tzapu, 2016-02-08:
i added a fix for that in the latest 0.9 i think, update WiFiManager and see if it still does it

fat-tire, 2016-09-06:
FWIW I'm seeing the same:

*WM: Configuring access point...
*WM: ESP123456  //changedthis
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

Then blinking led.  Guess I can try tracking it down but wanted to post  jic anyone else seeing it.