tzapu/WiFiManager #91: ESP does not remember configuration

Oripy, 2016-02-04:
Hello!

Thank you for this library, it is very useful!

I am running into an issue, I updated the library and now my ESP does not remember the configuration:
- I am able to configure using the portal.
- The Wifi connects to the chosen SSID (with the right password entered) just fine.
- But if I reboot the device, the configuration is lost and my device goes to access point mode again.

Note that it used to work fine when the configuration was stored on the EEPROM. Now that you removed this dependency, where is it stored?

tzapu, 2016-02-04:
hi,

the config is stored by the expressif sdk itself now (still in eeprom, but managed by the sdk)

what version of wifimanager are you using and of esp8266 core?

have you got any delay() in setup() ?

cheers

osos, 2016-02-04:
I have the same issue.

how is it handled by the expressif sdk? I do not see any call to any function for storage in the code.

Also is this only on version 2.1.0 of the esp8266 arduino or also on 2.0.0 ?

tzapu, 2016-02-04:
all versions of the esp8266 core handle the saving of ssid and pass in eeprom

do you have a resetSettings() or disconnect() anywhere in your code?

osos, 2016-02-04:
so it is handled by the WiFi-class ?

no I do not have any resetsettings() or disconnect() in the code.

tzapu, 2016-02-04:
well, lower level than any arduino stuff
i think set config saves it

tzapu, 2016-02-04:
guys, i ve just rememberd running into a similar behaviour at some point, 
in my case, if i had the Flash size of the module set to the wrong size, nothing would work.

Could you check you have the flash size set properly 512kB, 1MB, 4MB for your module, and that the module is indeed that size (example sketch esp8266/CheckFlashConfig)

Oripy, 2016-02-04:
Wait, I may have found why it behaves like this in my case.
My code do a "triple" connect:
- First try to connect to hardcoded SSID values ("factory" settings for internal tests)
- if failed try the previously entered values (if any)
- if failed again go to Access Point

I guess that my first try with my hardcoded values is erasing the previously entered values! If so, I guess I will have to branch from WiFiManager to reimplement EEPROM config saves or swap the first two tests.

I will try that the next time I have a chance to work on this project (maybe this week-end).

osos, 2016-02-04:
I am pretty sure I have chosen the right flash size: ESP-01 1MB (64SPIFS). I will check up and revert on this,

However, I am still interested in documentation on how the storage of the ssid/password storage is triggered and any limitations or requirements in SDK in this regard. Do you have a pointer to such documentation?

tzapu, 2016-02-04:
hi, 
but why not let wifimanager do all? 
autoConnect does all that and you don t need to do yourself anymore wifi begin, or anything like that...

if you do need to try first with factory settings as you say, i think there is a flag so you can tell it to not persist the values to the db 
do `WiFi.persistent(false)` before your begin, then try another `WiFi.persistent(true)` before wifimanager, if it doesn t connect... not sure if it will work though

tzapu, 2016-02-04:
@osos from the esp8266 sdk guide
7. wifi_station_set_config
Function:  
Set WiFi station configuration, and save it to flash
Note:
   •  This API can be called only if ESP8266 station is enabled.
   •  If wifi_station_set_config is called in user_init , there is no need to
          call wifi_station_connect after that, ESP8266 will connect to router
          automatically; otherwise, need wifi_station_connect to connect.
   •  In general, station_config.bssid_set need to be 0, otherwise it will
          check bssid which is the MAC address of AP.
   •  This configuration will be saved in flash system parameter area if
          changed.
Prototype: 
bool wifi_station_set_config (struct station_config *config)

Oripy, 2016-02-04:
Thank you for the info on Wifi.persistant(), I will try it, it seems to be just the thing I need.

> autoConnect does all that and you don t need to do yourself anymore wifi begin, or anything like that...
I do not understand your comment, is there a way to set a default value that is already defined just after flashing a brand new ESP8266?

tzapu, 2016-02-04:
sorry, i missed your factory defaults wish
to set your defaults you d need to do a begin the first time you run it, so you could do something like 
 wifi get ssid
if ssid is blank/null try connecting with your defaults
go through wait for connection results
then add all the simple autoConnect from wifi manager, it will see you are connected and return
if you are not connected it will start config portal
if you already have a ssid, then you must not be at first run

Oripy, 2016-02-04:
Thank you for your time! I will close this issue as soon as I checked that this was my problem.

tzapu, 2016-02-04:
cool , do let us know :)

osos, 2016-02-05:
I did use the right flash size. However, I did have some FS access which also failed.
After removing the FS code and chosing a diffrent SPIFS ratio, it works.

Maybe the SDKs continues save of the credentials damages some of the flash cells?

Anyway it works now, using WifiManager 0.9, ESP8266 2.0.0 and Arduino 1.6.7.

The sketch is a simple DHT22 waking up after deepsleep posting its data to thingspeak.com.

tzapu, 2016-02-05:
yeah, some weird stuff happening with spiffs/eeprom/credentails, i have seen some other weird behaviour cleared after a FS.format()

btw, could you let me know in a week or so if your deep sleeping sensor is still working?
i have one that after a few days, it goes through the motions alright, it s just the http request that is failing... based on newer sdk. another one that i have and is based on an older sdk, maybe 1.2 , has worked flawlessly for more than half a year...

PuceBaboon, 2016-02-05:
Interesting you should say that.  I have two battery-powered sensors which
I recently
updated (-not- using WiFiManager) using Arduino-ICE 1.6.7 with 2.1.0-rc2
and they've
both gone off line after maybe only five days, whereas before they'd been
running quite
happily for weeks (of course, the reason I updated them was because I'd
made some
changes too, so it's a bit early to point a finger at anything other than
my poor
programming).  :-)
I need to go and retrieve them and plug them into my laptop to see what
they're showing
on the serial port.

osos, 2016-02-08:
After getting wifi manager working with my dht connected esp-01, it just gone offline 6 hours ago.

Will se if I can work out the state when I get home.

osos, 2016-02-11:
@PuceBaboon, @tzapu regarding the DHT22 and other stop working see: https://github.com/esp8266/Arduino/issues/654 og https://github.com/esp8266/Arduino/issues/1594

Seems the new SDK for esp8266 does not allow "NoInterrupt", which is used inter alia in the DHT lib.
Interrupt-driven DHT lib can be found here: https://github.com/chaeplin/PietteTech_DHT-8266

tzapu, 2016-02-11:
ooo... what a great piece of intel.... thanks, i will try it in the next couple of days... would be great if this sorted it

Oripy, 2016-02-15:
Sorry for the late feedback.
I tried several things but I'm not good enough to understand what is going on. So I modified my requirements and the "factory settings" are accessed only when a button is pressed during boot. That solved my issue, but it is a workaround.

tzapu, 2016-02-15:
hi, you could just add this at the begging of your sketch , before any wifiManager stuff
```
if(!WiFi.SSID() || WiFi.SSID() == "") {
  WiFi.begin(default_ssid, default_pass);
}
```
(code not tested)

closed 2016-03-08 by tzapu

fvpalha, 2016-07-15:
I have the same problem. The new SDK (1.5.4) does not change the SSID and PASSWORD in STATION mode.

kadellabs, 2017-11-04:
hi how to get ssid and password of previously connected WIFI ? in esp8266-01

bhrigs, 2018-06-14:
Erasing the flash completely (you need to upload the blank_1MB.bin file to 0x000000, 0x100000, 0x200000 and 0x300000) as per the following link, solved it for me:
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

tablatronix, 2018-06-14:
Or use esptool..

Portia-Lin, 2019-07-14:
Hello. I can not solve this problem. I'm using a standard sketch "AutoConnectWithReset". After pressing the reset button, ESP8266 does not plug into a choosen network. It enters the AP mode. I add the output text from the serial port:
https://pastebin.com/JJj7at2j
And when I connect to an access point again there writes this:
[image]

Why ESP8266 does not reconnect after pressing reset button? 
I do not know what the reason is :с