tzapu/WiFiManager #92: setTimeout not work properly

chuxuanhy, 2016-02-04:
Hello,

I am using ESP12E with version 2.1. rc2 and wifimanager 0.9.
I set timeout to 180 seconds. Correct uid and password was saved. Wifi signal was also pretty good (above 80%). The ESP12E gave up quickly to try to connect to the Internet (about 3-4 seconds not as long as the timeout was set) at power on, then the config portal displayed in the wifi list. It was also randomly to make successful or fail connection at the time of powering on. I used AutoConnect, please check this!

tzapu, 2016-02-05:
hi, as it is, the timeout is from the moment wifimanager is intialised, so it would be 180 seconds from the moment you booted pretty much.

i am reconsidering this
cheers

chuxuanhy, 2016-02-06:
There must be an issue, I also tried increasing the timeout (to hour) but it helped nothing. ESP12E did not try to reconnect to the internet. It just gave up and started the config portal. 

I ran the OnDemandConfigPortal example, gave corrent credentials, had good wifi signal, but could not connect to the Internet. Please recheck the sketch!

tzapu, 2016-02-07:
hi, you may be misunderstanding the timeout parameter. 
what it is is a time it waits in the config portal, not how long it tries to connect.
when timeout is reached, WiFiManager terminates itself and returns control to your sketch.

chuxuanhy, 2016-02-07:
Oh, I misunderstood the timeout parameter completely! May you help me to modify the sketch to make it try to reconnect in an interval of time? The issue is that it randomly failed to connect to the internet at the time of powering on. Please help me!

tzapu, 2016-02-08:
hi, try the example AutoConnectWithTimeout.
If it doesn t manage to connect in 3 minutes, it will reset the module and try again.
Will that do it for you?

closed 2016-02-16 by tzapu