tzapu/WiFiManager #93: AutoConnectWithFSParameters Examples not working

ratzl, 2016-02-05:
Hi,
I've tried both examples for adding custom parameters, both are not working.
Custom parameters will not be saved, and ESP is not connected to AP.

I'm using core 2.1.0 and lib 0.9

*WM: Parameter
*WM: server
*WM: bla
*WM: Parameter
*WM: port
*WM: 8081
*WM: Parameter
*WM: blynk
*WM: YOUR_BLYNK_N
*WM: static ip
*WM: 10.0.1.56
*WM: static gateway
*WM: 10.0.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.56
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.

tzapu, 2016-02-05:
hi

that looks wrong

*WM: static ip
*WM: 100156
*WM: static gateway
*WM: 10011
*WM: static netmask
*WM: 2552552550

missing some dots…

ratzl, 2016-02-05:
sorry ???

tzapu, 2016-02-05:
100156 is not a valid ip

nor is 2552552550

ratzl, 2016-02-05:
did't you see te dots in your browser?

*WM: Parameter
*WM: server
*WM: bla
*WM: Parameter
*WM: port
*WM: 8081
*WM: Parameter
*WM: blynk
*WM: YOUR_BLYNK_N
*WM: static ip
*WM: 10.0.1.56
*WM: static gateway
*WM: 10.0.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.56
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.

ratzl, 2016-02-05:
btw, i have only edit the server,port and blynk parameter

tzapu, 2016-02-05:
no, something must of been off on the email formatting, weird.
in the examples, if it doesn t connect, it doesn t save, so that s why that is happening.
as for connecting, i don t know why it behaves like that, it s something amiss in newer sdk s...
check the following:
- no delay() in setup()
- arduino ide flash size set to proper module size
- do a FS.format(), and/or change spiff size beforehand.
- check one of the simple autoConnect samples see if they work
i hope some of those will sort it
cheers

ratzl, 2016-02-05:
same problem with simple AutoConnect ... not connecting

tzapu, 2016-02-08:
hi, did you check/try these
- no delay() in setup()
- arduino ide flash size set to proper module size
- do a FS.format(), and/or change spiff size beforehand.

ratzl, 2016-02-12:
hi,
after a long test period I've found out some strange behavior:

To change custom parameters I need the following steps:

-Enter Config Page
-Insert wrong ssid and password , --->save settings.
-Power cycle ESP Module
-Enter Config Page
-Insert correct ssid and password , and insert custom parameters--->save settings.

Now custom parameters are saved !

tzapu, 2016-02-16:
so, if you enter the correct ssid and pass the first time, the custom params are not saved?

closed 2018-01-26 by tablatronix