tzapu/WiFiManager #94: Wifi password should not be included in console debugging

vicnevicne, 2016-02-05:
Just a suggestion:
I was surprised to see my wifi password in clear text on the console. I know debugging can be turned off, but I think many people will leave it on, and just reading from TX and resetting the module would disclose the password.
I would suggest replacing it by stars *******, or better yet, make it optional by upgrading 
void WiFiManager::setDebugOutput(boolean debug)
to
void WiFiManager::setDebugOutput(int debugLevel)
where 
level=0 means off
level=1 means on with hidden password
level=2 means on including password
(these levels are just an example. More levels could be defined but the above values would keep 100% backwards compatibility while hiding the password unless the developer chooses so)

Keep on the good work !

tzapu, 2016-02-05:
hi,

you are correct, but, all you need to find out the password anyway is to write a blank sketch on it and call WiFi.psk()

not much in the way of protection anyway not showing it there

i ll think about it, and will do something about it probably when we ll move to the new ESP core debug functions

cheers

vicnevicne, 2016-02-05:
Thanks for your reply.
Regarding WiFi.psk(): that's true, but flashing requires a more complex setup than just connecting a console on the wire marked "Tx" :-)
Probably it would it good to think about it in the future indeed.
Anyway, your Manager works like a champ. Well done !

kentaylor, 2016-04-21:
I agree with tzapu that the password is so easily accessible by updating to a new sketch it doesn't seem worth having the option to hide it from the debug messages. Flashing a new sketch doesn't require any hardware, it can be done from the terminal program being used to read the debug messages. 

I'd expect in a finished product the debug messages would probably be turned off.

closed 2018-01-26 by tablatronix