tzapu/WiFiManager #95: Remove version warning in library.properties

vicnevicne, 2016-02-06:
README.md version history indicates that the current version is compatible with esp core 2.0.0 or newer, but the paragraph in library.properties still says **USE version 0.6 together with stable 2.0.0 ESP8266...**

tzapu, 2016-02-07:
hi,
does this actually show up anywhere?

vicnevicne, 2016-02-07:
Well it shows up right in the library manager of the Arduino IDE :-) :
[image]
Thanks for the fix.

tzapu, 2016-02-08:
jeez, now i feel stooopid ..:P
i looked for it after adding it and couldn t see it anywhere, must of taken a bit to update.. and when i looked again i obviously didn t notice it. it will probably dissapear once i post 0.10
thanks

closed 2016-02-08 by tzapu