tzapu/WiFiManager #96: Library Manager

pieman64, 2016-02-07:
tzapu when I request wifimanager in the library manager of the IDE it is offering me version 0.5.0 or 0.6.0. I notice you are up to version 0.9.0. Do I need to manually install your library if I require a version beyond 0.6.0?

tzapu, 2016-02-07:
i m seeing all of them .. maybe you need a newer arduino ide? 
[image]

vicnevicne, 2016-02-07:
If you're on Windows, there is a known bug in Arduino library manager up to 1.6.7 which prevents a refresh of the libraries. Probably that's what you are seeing.

To force a refresh, remove the following files:
C:\Users\<your_profile>\AppData\Local\Arduino15\library_index.json
and
C:\Users\<your_profile>\AppData\Local\Arduino15\library_index.json.tmp.gz
and restart the library manager.

The fix is in for 1.6.8. See https://github.com/arduino/Arduino/issues/4272

tzapu, 2016-02-08:
thanks @vicnevicne

domonetic, 2016-02-08:
Wifimanager will be updated in Arduino 1.6.8. As was advised at the time of the aggregate.

-----Mensaje original-----

Enviado el: ‎07/‎02/‎2016 11:47
Para: "tzapu/WiFiManager" <WiFiManager@noreply.github.com>
Asunto: Re: [WiFiManager] Library Manager (#96)

The fix is in for 1.6.8. See arduino/Arduino#4272

pieman64, 2016-02-08:
Thanks guys. I upgraded from 1.6.5. to 1.6.7 and still nothing beyond 0.6.0 available. Disliked 1.6.7 so went back to 1.6.5. and then 0.9.0 was available. Tried the first WiFiManager example on my Wemo mini and very impressed. Off to try the custom stuff now for Blynk auth. Is there anything WiFiManager related that I can't do with 1.6.5?

Humancell, 2016-02-08:
Just as a related comment ... the documentation says: Quick Start -> Installing -> Through Boards Manager ...

Shouldn't this really be: Quick Start -> Installing -> Through Manage Libraries ...

tzapu, 2016-02-08:
arduino ide version shouldn t matter

esp8266 core should be 2.1.0 (staging) to be able to set custom ip and save to FS

tzapu, 2016-02-09:
@Humancell correct, thanks

pieman64, 2016-02-10:
@tzapu I have now moved on to AutoConnectWithFSParameters and it connects my Wemos to the router as expected. I'm a bit confused on the flags to set if I want to save the settings (SSID PWD and BLYNK_AUTH).

Looks like it is the bool shouldSaveConfig. After the bool declaration as false there is the function saveConfigCallback. I have tried changing the declaration to true and false but it doesn't seem to give me the results I expected. Is it just this declaration that covers saving the details or is there another flag elsewhere?

With the flag as true if I reboot the Wemos I get the following in Serial Monitor

```
Mounting FS...
Mounted file system
*WM: Adding parameter
*WM: blynk
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.10.111
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.10.111
Connected Central Heating System :)
saving config
{"blynk_token":"YOUR_BLYNK_TOKEN"}local ip
192.168.10.111
```

I'm not convinced I have everything set up correctly because of the settings invalidated comment and YOUR_BLYNK_TOKEN where I perhaps expect the actual token.

The token is 32 characters and I guess you need 1 extra place in the array for the terminator but you show an array is 34 characters including the terminator.

tzapu, 2016-02-10:
hi

remove wifiManager.resetSettings();

other than that, it does show like it works. don t get hung on array length, should be +1

cheers

pieman64, 2016-02-10:
Just found resetSettings, will give it a go and thanks for all your help. Loving WiFiManager.

pieman64, 2016-02-10:
Working like a dream now. When I turn off my secondary router it brings up the AP configuration and when I switch the router back on it signs in automatically. Will check out your "LED" now before incorporating into our Blynk sketches.

One thing I have noticed with the ESP's is that they don't seem to give a true representation of the wireless signal strengths. I have the main router in an adjoining room and even though it is about 70% signal strength it very rarely shows up with a scan. Whereas other routers in the building with strength's just above your 8% minimum do show up. I don't have any problem connecting the 70% strength router and it the SSID is not hidden but it generally fails to appear on a scan. Have you noticed this?

tzapu, 2016-02-10:
hi, there s a fix in the latest esp8266 core , github version for scans.

it should be included in the next staging probably

pieman64, 2016-02-10:
Thanks for the scan fix which I think I have completed. It seems better now but I'm not 100% convinced I have it right. One router is 6" from my PC and reads 100%, the main router next door is 76% but when I keep scanning over and over again I see one or the other plus lots of 20% to 50% routers but never both of mine together. That suggests I might still be missing the first index in the scan or ESP's are a little crazy.

pieman64, 2016-02-10:
After around 70 or 80 scans I finally see both my routers. Strange little things these ESP's.
[image]

pieman64, 2016-02-10:
@tzapu me again. Added WiFiManager to my Blynk sketch but I'm not getting the correct Blynk token.
For now I am trying **Blynk.begin(blynk_token, "Office", "XXXXXX" );** until I know where to pick up the ssid and pwd or login to Blynk with just the token as I'm already connected to the router with your part of the sketch.
The best I can do can get is:

```
[498289] Connecting to cloud.blynk.cc:8442
[498574] <msg 2,1,31
<518430403ce640ef8xxxxxxxxxxxxxx
[498777] >msg 0,1,9
[498777] Invalid auth token
[503778] Connecting to cloud.blynk.cc:8442
[504103] <msg 2,1,31
<518430403ce640ef8xxxxxxxxxxxxxx
[504307] >msg 0,1,9
[504307] Invalid auth token
[498289] Connecting to cloud.blynk.cc:8442
[498574] <msg 2,1,31
<518430403ce640ef8xxxxxxxxxxxxxxx
[498777] >msg 0,1,9
[498777] Invalid auth token
[503778] Connecting to cloud.blynk.cc:8442
[504103] <msg 2,1,31
<518430403ce640ef8exxxxxxxxxxxxxxx
[504307] >msg 0,1,9
[504307] Invalid auth token
```

Note the leading **<** in the token. Any ideas?

pieman64, 2016-02-10:
Just to add the far right digit is missing in the token, so correct length of 32 characters but with the trailing **<** error.

tzapu, 2016-02-10:
hi,

i ll try to add a blynk example sketch tomorrow if i get a chance, to use as a starting point.

does your config json look ok?

pieman64, 2016-02-10:
Ok so the leading < is just the Blynk 'prompt'.

Changed:

WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);
to:
WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 33);
and I'm now connected to the Blynk server.

pieman64, 2016-02-11:
@tzapu I'm not quite there with WiFiManager and Blynk.
If I reset the settings and enter the details in the web portal I can connect to Blynk but if I restart the Wemos and try to connect to the same router Blynk isn't accepting my token.

Can you just clarify the bool settings to ensure I am saving the credentials correctly and if I need to make any other changes having set the parameter to a length of 33 (from 32)?

tzapu, 2016-02-11:
hi,

in the sample code, the bool setting is just used lower down the script to check if it was set, and then save configuration.

that only happens after a sucessful config/connect

does that make it clearer?

pieman64, 2016-02-11:
So the declaration is made as false and then the function sets it as true, right?
That is how I currently have it.
Any thoughts on the token length as I have seen others with 33 characters in their sketches?

tzapu, 2016-02-11:
hi yes, correct

it needs to be 33 or more to include the termination character.

cheers

scropion86, 2016-02-11:
@tzapu Please in your Blynk sample code if you can add a V pin that will again raises AP mode for WIFI and parameters configuration. is that possible instead of the physical button connected to actual Pin .

pieman64, 2016-02-11:
Worked out what my problem was. I still has SPIFFS.format() in my sketch, all good now.

tzapu, 2016-02-29:
@scropion86 if you want, please open a new issue with this:
"Please in your Blynk sample code if you can add a V pin that will again raises AP mode for WIFI and parameters configuration. is that possible instead of the physical button connected to actual Pin ."

cheers

closed 2016-02-29 by tzapu