tzapu/WiFiManager #97: Access SSID and Password after using On Demand Portal?

Humancell, 2016-02-09:
Like everyone else says ... awesome library!  It worked on the first try.

The way we are writing our app, we're doing our own WiFi connection handling, but want to use your captive portal/On Demand Portal for the configuration.  I put a loop in the Setup() that give the user 5-6 seconds to press a button that tells us to call wifiManager.startConfigPortal("Our AP SSID") ... all of this works fine.

If the connection - after configuration - was successful, how can I retrieve the SSID and Password so that we can store it in our other configuration files?

After the user does the Captive Portal configuration, we want to store the configured credentials in our own config files, and then reset the board no matter what.  When it reboots, and they do not press the button, our own WiFi connection code will then retrieve the credentials and initiate the WiFi connection.

We don't want to use autoConnect() since we do not want the library to go into Captive Portal mode ever ... unless the user pressed the button during that initial power up 5 seconds.

Does that make sense?

Humancell, 2016-02-09:
It does look like you once provided:

    //String        getSSID();
    //String        getPassword();

But then you commented these public methods out?

tzapu, 2016-02-09:
yeah, all they did was call
WiFi.psk() and WiFi.SSID()

Btw, if i understand correctly, you don t need to save them, in your connection management code just put 
WiFi.begin() with no params and it will use the last saved values.

I haven't tested, but you could try with 
```
 //exit after config instead of connecting
  wifiManager.setBreakAfterConfig(true);
```
to exit and trigger a reset. so, semi pseudo code
if trigger pressed 
```
wifiManager.setBreakAfterConfig(true);
wifiManager.startConfigPortal
```
when config portal returns 
```
ESP.reset()
```
if no trigger button was pressed
```
WiFi.begin()
```

that should do all you mentioned

Humancell, 2016-02-10:
So close!!  Thank you for the suggestions ... I'm running into two issues, and am curious if these might be bugs ...

First, I'm following your steps above, and it works ... BUT ... when I call the portal and then the reset I get the following output:

```
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

... and then it just hangs forever.

THEN, if I manually reset the ESP, it DOES work ... and connects ok.  But then at the end of my processing, I call ESP.reset() again ... and it's almost like the PSK is wiped out.  When the ESP restarts I can print the WiFi.SSID() and it's still there ... but it will no longer connect until I re-configure it.  I can't find a way to print the stored PSK ... it seems that WiFi.psk() no longer exists?

Humancell, 2016-02-10:
I've been sitting here testing and testing ... and the wdt reset issue has simply gone away ... I'm not sure how.

I'm still looking at why the ESP will not reconnect to one of my access points on subsequent ESP.reset() calls ...

tzapu, 2016-02-23:
hi, WiFi.psk(); should really be there ...
could you have an older version of the esp8266 core for arduino?

Humancell, 2016-03-15:
Ok ... this issue is resolved.

1. Yes. .. I did get the WiFi.psk() to work ... so that is good.

2. I found out the reset issue finally ... it's related to another issue that is talked about.  For some reason the first time the ESP resets after being flashed, it doesn't seem to reset correctly.  It works on all subsequent resets!

This is working great!

Can't wait to get the ability to customize the CSS ... I don't want to modify the pages yet so I don't conflict with your changes.  Where do I donate?

closed 2016-03-15 by Humancell

tzapu, 2016-03-15:
hi, in the latest github version there is the ability of adding custom css, you can use that now

didn t think about donations, would people really do that?

Humancell, 2016-03-15:
Oh excellent!  I'll check out the CSS!  I was also thinking of having my team see about adding a "spinny" busy icons when you touch a button ...

Donations?  I know we would!  You've done a great job on this library, and we're about to include it in a part of our open source release of our ESProto sensor prototyping platform. :-)

Humancell, 2016-03-15:
BTW ... is there any doc/suggestions on adding custom .css?

tzapu, 2016-03-16:
hi, you can use it like this:
```
wifiManager.setCustomHeadElement("<style>html{filter: invert(100%); -webkit-filter: invert(100%);}</style>");
```
docs will be added when i release 0.11

i have never taken contributions so i have no idea how it s best done

i ve setup a patreon page, which is apparently the new "hip" way of doing this from what people say on the interwebs.

i have to stress that I will continue development on this, indifferent of contributions, and that if anyone decides to contribute, no amount is too small, everything goes the way of supporting my programming and new found electronics habit :P .

https://www.patreon.com/tzapulica

shakirhussain78652, 2021-02-03:
Hello,
I am new here with not much basic knowledge. I am trying to use this library Auto Connect Non Blocking wifi manager. I need the network Wifi SSID and the password entered by the user at the configuration portal to be stored in variable to be used in the REMOTE XY app.
Please provide me with the code.

// RemoteXY connection settings 
#define REMOTEXY_WIFI_SSID "hello"  // this is where i want to use the ssid 
#define REMOTEXY_WIFI_PASSWORD "123456789" //and password
#define REMOTEXY_CLOUD_SERVER "cloud.remotexy.com"
#define REMOTEXY_CLOUD_PORT 6376