tzapu/WiFiManager #99: Crash after selecting network

fmgomes, 2016-02-09:
I am using the simplest example, Autoconnect. I was testing it using different browsers and devices (PC, Smartphone, etc.) initially it worked, and after some tests the result I have is always the same, when I select the network on the webpage, it tries to connect to it and crashes. It happens with different networks. This is the log:
...
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...

Exception (9):
epc1=0x40105479 epc2=0x00000000 epc3=0x00000000 excvaddr=0x2087dc3f depc=0x00000000

ctx: cont 
sp: 3fff04a0 end: 3fff0a00 offset: 01a0

>>>stack>>>
3fff0640:  00000000 4000444e 3fff2ab4 3fff2ed8  
3fff0650:  3fff3658 00000601 40243370 401056b1  
3fff0660:  000003fd 00001000 00000484 003ff000  
3fff0670:  00000484 401056de 000003fd 003ff000  
3fff0680:  40229545 3fff06b0 0000001c 003ff000  
3fff0690:  401059ae 4022962d 3fff2e10 000003ff  
3fff06a0:  000003fd 3fff07b7 3fff2e10 000003fd  
3fff06b0:  ffffff00 55aa55aa 00000042 0000001c  
3fff06c0:  0000001c 000000d9 000000d8 000003ff  
3fff06d0:  40229a1c 3fff2e10 3fff2e10 000000ff  
3fff06e0:  00000001 3fff07d7 40229b77 00000008  
3fff06f0:  3fff2e10 000000ff 3fff07b7 00000000  
3fff0700:  3fff2ed0 3fff0818 00000001 40229c04  
3fff0710:  3fff07b7 3fff2e10 00000000 00000004  
3fff0720:  3fff07d7 3fff9148 3fff2e10 00000000  
3fff0730:  40229c40 3fff1ee0 3fff1e80 402024d6  
3fff0740:  40202aee 3fff1e80 3fff1ee0 40202a2e  
3fff0750:  3fff0a00 00000001 3ffe8849 40208818  
3fff0760:  3fff0a4c 00000004 3ffe8cc0 3ffef9cc  
3fff0770:  3fffda00 0000005e 3fff0a4c 402082fd  
3fff0780:  3ffe8848 40241604 3fff0a4c 3ffef9cc  
3fff0790:  3fff0a4c 0000005d 3fff0a4c 402088f0  
3fff07a0:  3fffdad0 40241604 3fff0a4c 40208911  
3fff07b0:  ffffff00 78ffffff 2d6f7470 0074656e  
3fff07c0:  00000020 00000000 00000010 3ffe884a  
3fff07d0:  3fff0a4c 00000001 3ffe8849 402087dc  
3fff07e0:  3fff0a4c 00000004 3ffe8cc0 00000000  
3fff07f0:  3fff08bc 0000001c 3fff0a4c 402082c1  
3fff0800:  3ffe8848 40241658 3fff0a4c 00000000  
3fff0810:  3fff0a4c 0000001b 3fff0a4c 402088b4  
3fff0820:  3ffef860 00000000 00000001 402088d5  
3fff0830:  3fff1ed8 40241658 3fff0a4c 3ffe8eac  
3fff0840:  3fff08bc 3ffef860 3fff0930 402064b4  
3fff0850:  3ffef860 00000014 3fff0a4c 4020921a  
3fff0860:  3fff1490 00000000 3fff08b0 40209272  
3fff0870:  3fff0a4c 00000013 3fff08b0 402092a2  
3fff0880:  3fff08b0 00000000 3fff08b0 402093a4  
3fff0890:  402012e6 4024168c 3fff0930 00000000  
3fff08a0:  3ffef860 00000000 3fff0930 402069d9  
3fff08b0:  3fff1ee0 0000000f 00000000 3fff1e80  
3fff08c0:  0000000f 00000008 3fff08fc 402092f0  
3fff08d0:  3ffe8eac 402416b0 3fff0a4c 00000000  
3fff08e0:  00000000 3ffe85e0 3fff0930 40206b09  
3fff08f0:  00000000 00000000 00000000 00000000  
3fff0900:  00000000 00000000 feefeffe feefeffe  
3fff0910:  feefeffe feefeffe 3fff0930 3ffef9cc  
3fff0920:  3fffdad0 00000000 3fff0a4c 40202015  
3fff0930:  3fff14a8 3fff1508 3ffe85e0 00000000  
3fff0940:  3fff1478 0000000f 00000008 3fff1490  
3fff0950:  0000000f 00000000 00000000 00000190  
3fff0960:  3ffe9148 00000000 3ffe9148 00000000  
3fff0970:  3ffe9148 00000000 3ffe9148 00000000  
3fff0980:  3ffe9148 00000000 3ffe9148 00000000  
3fff0990:  00000000 ffffffff feef0000 00000000  
3fff09a0:  fe010035 00000000 00000000 feefeffe  
3fff09b0:  feefeffe feefeffe feefeffe feefeffe  
3fff09c0:  feefeffe feefeffe feefeffe feefeffe  
3fff09d0:  feefeffe feefeffe feefeffe feefeffe  
3fff09e0:  feefeffe feefeffe 3ffef9c4 40207a18  
3fff09f0:  feefeffe feefeffe 3ffef9e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

Any idea what could have been the cause and how to correct it?

Thanks!

Fernando

tzapu, 2016-02-10:
hi

try with another module if you can and with another power source.

connection issues like that are usually caused by inadequate power supply

cheers

fmgomes, 2016-02-10:
Hi, thanks for your reply! I think that it was something that occured in one of the configuration tests, because I've been able to change the configuration a few times using the WiFi Manager without any issue, and after some point I always have the crash when it tries to connect to the network. This issue was solved when I wrote a new program to the module. The module was a NodeMCU, and it never crashed before, with lots of different sketches running on it (I previously had lots of crashes using the ESP-01 due to power and module layout problems, then I changed to the Olimex boards, ESP-12 modules  and NodeMCU modules and never found that problem again). One thing that might have changed during the test (but I'm not sure it was the cause) was that in the initial tests I was using the smartphone to configure the WiFi, and then I changed to my PC, that had the brower already open and might have some of the webpages trying to get data from the server, when I changed to use the NodeMCU as an AP, the NodeMCU started getting these requests, not only a single page request to make the configuration. Do you think that it might have this secondary effect? I will test it again trying to reproduce the problem, but probabbly only next weekend.

Best regards

Fernando

closed 2016-02-10 by fmgomes