tzapu/WiFiManager #100: Trying to connect to password less ap uses old settings

solexious, 2016-02-12:
Hello,

I'm currently troubleshooting trying to connect to a ap, I removed it's password to fault find why it was failing to connect. I noticed that if I entered the ssid but no password into the wifimanager connection form the serial output shows it just tries to connect to my last known good wifi settings:

*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
connected...yeey :)

Scanning the new network it should be connecting to now, and the old network that it was last saved to shows it is now connected to the old network.

tzapu, 2016-02-12:
hi, this was a bug fixed in 0.8 or 0.9, please update WiFiManager

solexious, 2016-02-12:
updated from 0.8 to 0.9 with the same issue still

tzapu, 2016-02-12:
in the latest wifimanger the condition is `  if (ssid != "") ` so it shouldn t really matter if the password is empty or not
are you still getting this in the log
```
*WM: Using last saved values, should be faster
```
after 
```
*WM: Sent wifi save page
```

solexious, 2016-02-12:
Tried it again now and its now working, odd.

Thanks though!

closed 2016-02-12 by solexious