tzapu/WiFiManager #102: Testing connection to new AP gives up easily

solexious, 2016-02-12:
I'm trying to connect to my tp-link tl-wr702n router through wifimanager.

Currently when it tries to connect it will fail out very quickly and not connect, where as it will connect fine to my wifi router in the other room fine.

I've been running my project fine with this module and wifi ap just calling wifi connect normally so I wonder if wifimanager is just timing out trying to connect really quickly, or needs to try multiple times?

I can't currently see any settings to tell it how hard to try to connect to a new network.

solexious, 2016-02-12:
I've noticed the connection error is always that the wifi is idle?

*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Handle root

tzapu, 2016-02-12:
hi, have you got the flash size under tools set to the same size your esp module has?

i have seen this happen at times with the wrong flash size set, also some weird combination of delays in the sketch cause it not to connect and fail immediately with 0

unfortunatelly this is quite random, i can t replicate it

tzapu, 2016-02-12:
i can t try to connect harder or longer, the library is using the exact same functions you and everyone else is using to connect, but due to mode switching from AP to STA weirdness like this seems to happen

solexious, 2016-02-12:
My details on the info page are:

Flash Chip ID
1458400
Flash Size
4194304 bytes

And I'm using "4M (1M SPIFFS)"

Is it possible to get it to try connecting a few times?

tzapu, 2016-02-16:
hi, if you can you should update to the latest github version of the library and check the new info page, there s an ide flash size and a chip flash size that should be equal

i have also added a new option to WiFiManger to set a connect timeout. 
in order to use this you d need tocall 
wifiManager.setConnectTimeout(30); 
before autoConnect. parameter is time in seconds to wait before bailing on connection.
i am very interested to see if it helps at all with your issue

solexious, 2016-02-16:
Thanks for making the changes. I've updated to the latest git version.

Info page tells me:
```
Flash Chip ID
1458400
IDE Flash Size
4194304 bytes
Real Flash Size
4194304 bytes
```

I added the 30 second timeout for trying to connect, I had it connect 1 out of 10 tries. Also tried setting the timeout to 60. Timed the wait and its waiting the amount of time I set t to.

I once also had it return connection result 4, but other than that the failures were all 0.

```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Waiting for connection result with timed out
*WM: Connection timed out
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
```

tzapu, 2016-02-16:
that s a pitty...
i am only using the standard functions so I do not know why it would connect ok for you just callign wifi connect and not through the lib...
can you try a simple sketch to find out if that s still the case, if it still connects ok from the first attempt?

tzapu, 2016-02-16:
if you don t use WiFiManager, in a test sketch, does everything work ok? Does your module see and connect ok to the router causing you issues?

solexious, 2016-02-16:
Given the basic WifiClient esp8266 sketch another try, connected fine first time each try, swapped back to wifimanager and had the same issue as before.

tzapu, 2016-02-17:
ok, thanks for letting me know.

given that i only use the standard functions, i ll think what the differences are (of calling order, as that s the only thing) and try to build a sample sketch we can use to debug and send to esp8266 core so they can research the possible bug there.

will you be able to test stuff?

solexious, 2016-02-17:
More than happy to. :)

tzapu, 2016-02-23:
hi,

sorry for the delay

could you give this a spin?
https://gist.github.com/tzapu/5b13866aaa584b32074a

it basically tries to duplicate as much as possible from the WiFiManager flow, without all the nice things, to see if we can find where the issue lies...

solexious, 2016-02-23:
I've tested that script 10 times, all 10 it worked correctly and connected fine showing the following serial output:

```
Start
1
3
Connected
```

tzapu, 2016-03-05:
i ve added a new fix to WiFiManager that tries to avoid a race condition that i found. i don t think it might be useful in your case, but worth a try nontheless...

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

solexious, 2016-03-10:
Apologies for the delay.

I've  just done a git pull and tested it with my module.

It now works perfectly with the examples in the library. I've tried it 10 times with 2 different AP's that the esp is trying to connect to (that it was failing on before) and it connected first time everytime.

Thanks so much for fixing it,

Charles

tzapu, 2016-03-10:
that is wonderful news
i ll give it a bit more time so if there s anything wrong people can report it, then will release it on lib manager

wvdv2002, 2016-03-15:
With the latest commit the ability to make a connection has been greatly improved. 
Before a lot of times it took a few restarts to get the connection up and running. Now it directly connects every time. Tested on Android hotspot and KPN Experia Box.

Thanks for solving this bug,
Wouter

tzapu, 2016-03-15:
good to know, thank you. i ll probably release it soon then

cheers

closed 2018-01-26 by tablatronix