tzapu/WiFiManager #103: Quit WiFiManager

pieman64, 2016-02-15:
@tzapu I have added a Quit option to your WiFiManager menu has shown below.
[image]

The menu option works fine except I don't know how to leave WiFiManager and return to the next line of my sketch. I currently have the following in your cpp file:

```
/** Handle the quit page */
void WiFiManager::handleQuit() {
  DEBUG_WM(F("Quit"));

  String page = FPSTR(HTTP_HEAD);
  page.replace("{v}", "Info");
  page += FPSTR(HTTP_SCRIPT);
  page += FPSTR(HTTP_STYLE);
  page += FPSTR(HTTP_HEAD_END);
  page += F("Join new AP ESP-XXXXX.");
  page += FPSTR(HTTP_END);
  server->send(200, "text/html", page);

  DEBUG_WM(F("Sent quit page"));
  delay(5000);
}
```
What extra code would I require to stop the AP created by WiFiManager and return to my sketch?

tzapu, 2016-02-15:
first thought, if you set the timeout to 1, it will exit immediatelly

tzapu, 2016-02-15:
another option would be to set connect to true and break after config to true, that would try to connect as well to last settings i think and then exit

pieman64, 2016-02-16:
@tzapu issue 86 regarding WiFiManager and webserver as identified by @knnniggett at https://github.com/tzapu/WiFiManager/issues/86 was also the problem I came up against yesterday.

I have a standard OTA webserver sketch that allows me to upload new firmware to my Wemos D1 Mini and within the sketch I have calls to WiFiManager. This is all tied to Blynk access so I want to be able to press one button in the Android app for firmware updates and another for accessing WiFiManager with additional paramaters (Blynk token and firmware revision number).

I spent all day yesterday trying to get them to work together but the best I could do was one or the other, not both. I will look through the sktech examples provided by the two of you and see if I have any success.

tzapu, 2016-02-16:
when wifi manager starts, it s spinning up the AP as well, which might drop the connection to the client and do all kinds of other bad stuff to the things in your sketch...
this was meant to start only when you can t connect to wifi - hence wifimanager, then we bolted on custom params and they would indeed benefit from being able to start the config portal without all the other crap :)
i have posted a http server example on a gist around here, do you have it? that should be a good starting point for a webserver with wifimanager

pieman64, 2016-02-16:
Do you mean this https://gist.github.com/tzapu/ecc0759829d30d5a6152 @tzapu?

tzapu, 2016-02-16:
that s the one

knnniggett, 2016-02-16:
Sorry guys, I've been focusing on another project as of late. I have not tried @tzapu 's gist yet, but it is on my to-do list.

closed 2018-01-26 by tablatronix