tzapu/WiFiManager #104: Working while not connected to Wifi

elieyal, 2016-02-16:
@tzapu, first, same as everyone, I think this library is amazing....
I have some flows I'm trying to do, maybe you can help me think on how to integrate this library into my flows:
1. The first time installation of the board in a new place where there is a new WiFi network
2. Letting the board continue to work as usual (inputs and outputs) while it's not yet connected to a network
3. Working as usual while WiFi is lost and resuming connection when the WiFi is again available
4. Changing the WiFi network on demand.

the idea behind these flows is very simple:
1. Configure and test
2. Move to a new place to do the actual work - reconfigure on site
3. make sure it's working no matter what - and stable.
4. Resume connectivity if lost.

Now the first flow is simple from your examples, but how can the board continue to work (with my board, gpio0 is an input button and gpio2 is output according to the button) if there is no WiFi configured?

What will happen if the WiFi is lost at some point? I have this in my current board:
//set IP if not correct
    IPAddress ipadd = WiFi.localIP();
    if( ipadd!= ip) {
        WiFi.config(ip, gateway, subnet);
    }
    // Connect to WiFi network
    WiFi.begin(ssid, password);
    counter = 0;
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        override_check();
        delay(250);
        
    }
    Serial.println("");
    Serial.println("WiFi connected");
    // Start the server
    server.begin()

Unfortunately, in your example you are using the ESP8266WebServer that I'm not so fund of, so i cant use your example that much (why use it when ESP8266WiFi is much more simpler?).

And last, you wrote in the documentation NOT to use both on demand and the autoConnect. so my question is how do I setup my board in one place and than move it to another place to work? how can I change the settings on demand but after the first initialization?

Thank you so much for the help!!

pieman64, 2016-02-16:
@elieyal when you move to a new place WiFiManager will automatically take care of things for you i.e. the SSID and pwd of the AP will not be stored in your ESP so it will bring up the autoConnect for you to configure the new location.

elieyal, 2016-02-16:
Right.
This is when I use the AutoConnect. But what will happen if my WiFi network will be gone and the board will restart? In this case, the AutoConnect will be brought up and my board will be on-hold until it's re-configured.
In my case, I don't have one board, i have 20 of them :-)
Re-configuring all of them is a problem.
I cant use the AutoConnect and OnDemand in the same board so it's a problem.
my flow should be like this:
1. if there is no SSID - bring up the portal for configuration (and in the mean time wait for gpio inputs and act when button pressed)
2. if ssid exists - try to connect to it. use it only. while trying to connect - continue to wait for gpio inputs
3. if a button is pressed for more than 5 seconds - bring up the portal for Wifi change (but continue to wait for gpio inputs)

does this make sense? maybe i should use only the onDemand, but how do i know that the SSID is stored?

tzapu, 2016-02-16:
after the initial setup, WiFiManager no longer interacts with your module.
if the connection comes and goes at least as far as i know, the default behaviour of the esp is to reconnect
you can use just on demand and a couple of ifs to duplicate the functionality of autoConnect and to have your sketch working , etc. you do not need to know what the ssid is
- when sketch starts issue WiFi.begin()
- if this fails (wait for connection result) or in your case you can check if WiFi.SSID() is empty, start on demand portal

while serving the config portal, the sketch is blocked. this is unlikely to change at any point due to how it all links together.

tablatronix, 2016-02-18:
Can you handle connection failures on your own and only start up

startConfigPortal() in a special mode or button press and not use autoconnect at all. ( the esp8266 will do its own autoconnect regardless, so thats fine )

otherwise just use your code as usual ?

Or are you wanting the portal to always be available, and also run your loops ?
Like STA + AP and always allow settings changes ?

tzapu, 2016-02-19:
```
while (WiFi.status() != WL_CONNECTED)
```
this seems to be exactly what is in the official WiFI.waitForConnectResult() btw

tzapu, 2016-02-19:
sorry, i stand corrected, the loop is actually 
```
uint8_t ESP8266WiFiSTAClass::waitForConnectResult() {
    //1 and 3 have STA enabled
    if((wifi_get_opmode() & 1) == 0) {
        return WL_DISCONNECTED;
    }
    while(status() == WL_DISCONNECTED) {
        delay(100);
    }
    return status();
}
```

kentaylor, 2016-05-02:
@tzapu  I think the function at https://github.com/kentaylor/WiFiManager/blob/master/WiFiManager.cpp#L293 does not go into an infinite loop because the espressif sdk will only return WL_DISCONNECTED for a limited time when in station mode.

@elieyal The example at https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino will do most of your scenario. To run that example you need the version of WiFiManager at https://github.com/kentaylor/WiFiManager/ . 

You asked for:-
"1. if there is no SSID - bring up the portal for configuration (and in the mean time wait for gpio inputs and act when button pressed)"

It will bring up the portal for configuration but ignore gpio inputs.

"2. if ssid exists - try to connect to it. use it only. while trying to connect - continue to wait for gpio inputs"

It does exactly that.

"3. if a button is pressed for more than 5 seconds - bring up the portal for Wifi change (but continue to wait for gpio inputs)"

The example brings up the portal immediately rather than waiting 5 seconds but it is easily changed to wait 5 secs. It will ignore GPIO inputs when the portal is open.

Opening the configuration portal is rare and it is human initiated so most applications should be OK with being non functional while WiFi is being configured. 

Also in your original scenario you wanted to take the devices to a second location with different WiFi credentials after testing them at a first location. With the WiFi Manager version at https://github.com/kentaylor/WiFiManager/ you can set the WiFi credentials of the second location while still at the first after testing. You will lose WiFi access at the first but will be able to just switch on the device at the second and it will work.

closed 2018-01-26 by tablatronix