tzapu/WiFiManager #105: startConfigPortal no connect after WEBconfig

Pfannex, 2016-02-16:
Hi, great stuff!

I got some problems to connect after using "startConfigPortal", "AutoConnectAP" works fine.

The code did not connect at this point in WiFiManager.cpp:
```
int WiFiManager::connectWifi(String ssid, String pass) {
..
..
..
  //check if we have ssid and pass and force those, if not, try with last saved values
  if (ssid != "") {
    DEBUG_WM("Using given values:");
    DEBUG_WM(ssid);
    DEBUG_WM(pass);
    WiFi.begin(ssid.c_str(), pass.c_str());

```

ssid and pass are given and were printed right.

but connection result is alwasy 0
```
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Handle root
```

Greatings from Hamburg
Pf@nne

tzapu, 2016-02-16:
hi, there seem to be some issues with connection simply not working until after a reset with no apprent cause yet.
does yours connect after a reset?
is your flash size set correctly in arduino IDE ?

Pfannex, 2016-02-16:
Hi,

ESP8266 12e -> 4M(1M)
Resetting will have no effect, no connection

elieyal, 2016-02-16:
Hi,
I seem to have the same problem.
I have the startConfigPortal in a function that is called if the SSID is blank in the setup or when pressing a button for 5 seconds.
The calling from the setup is working fine but when calling it after the button push it wont load the portal.
The function i wrote is very simple:
void reset_settings (void){
  //reset settings - for testing
  //WiFi.disconnect(true);
  WiFiManager wifiManager;
  wifiManager.setSTAStaticIPConfig(ip, gateway, subnet);
  if (!wifiManager.startConfigPortal("OnDemandAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  //ESP.reset();
}

the output:
GPIO0 was pushed
now change WiFi network
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: OnDemandAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

I get the AP and can connect to to it. The Portal wont load. Not as a pop up like the AutoConfig and not when trying to connect to http://192.168.4.1 from browser.

@tzapu, side question, why in the onDemand the configPortal wont pop up like in the autoConfig?

Pfannex, 2016-02-16:
The reason why I want to do that ist to write a basic library to connect both Wifi and MQTT. 
Maybe this should be a function within the WiFiManager Class by using a Callback if external parameters (MQTT connection) is successful.

[image]

tzapu, 2016-02-23:
@Pfannex sorry, but wifimanger should be flexible, including mqtt connection code for people not even using it won t really be any good.
it has sufficient callbacks and features for everyone to be able to include whatever other packages they want though...
also, there are a couple of reasons why it should not monitor connection either
- it would need to be kept in memory at all times (memory is at a premium) it would also keep a web server and dns server locked
- at least from what i saw, modules seem to reconnect after wifi goes away and comes back ( i did not test, just observation of the modules i have around, so might be quite wrong) . it would be quite a bad idea to spin up the config portal when wifi goes away, as it would reconnect anyway when wifi comes back up

@elieyal have you got any webserver code in your main sketch?

elieyal, 2016-02-23:
I fully agree with @tzapu. The wifimanager should not monitor the WiFi and this should be in the sketch if needed. I don't have a webserver code in my sketch but i do perform a check of the WiFi and if lost it will try to reconnect. also by doing this, I can control the behavior of the module and let it do things even if there is no WiFi connected - and this is the main reason why the wifimanager should not monitor the WiFi.
i'm sure there are other examples that people would like to monitor and control the WiFi on their own and not be locked by a package to do some functionality.
I do think that there should be another callback that can be enabled and disabled to allow the user to add his own function (or set of commands) while waiting for the WiFi to be connected if the OnDemand or autoConnect functions are called.

I have solved my issues by only using the OnDemand function. i'm checking if there is a stored SSID at the beginning, if there is not, the OnDemand is invoked. If there is a SSID it will try to connect to it and will not let try anything else. If there is a need to change, a button should be pressed for 5 seconds and than it simply reset the WiFi settings (delete the SSID) and reset the module. after restart it checks again the SSID and since there is non in the memory it will invoke the OnDemand again.
I have a function that checks for the WiFi connection and tries to connect back until it gets connected.

tzapu, 2016-02-23:
@elieyal good idea about loop callback, i think i should add that in..

kentaylor, 2016-04-22:
@tzapu said "it would be quite a bad idea to spin up the config portal when wifi goes away, as it would reconnect anyway when wifi comes back up". 

I'd agree with that but the same argument applies at startup as WiFi networks are flaky, yet at startup that is exactly what autoconnect does.

closed 2018-01-26 by tablatronix