tzapu/WiFiManager #106: Blynk and V pins

scropion86, 2016-02-16:
it's a question not an issue actually

1. i'am still not able to make it working with Blynk
2. is it possible to reset the ESP and after that to be in AP mode and showing the wifimanager page using a Virtual pin in Blynk.

thanks.

tzapu, 2016-02-16:
hi, you could do it by issuing a `wifiManager.resetSettings()` from the virtual pin and then with a delay `ESP.reset() or ESP.restart()`
you will of course loose ssid/password so would have to enter them again. If you have other settings saved to FS those will be ok

tzapu, 2016-02-16:
see here for a blynk sample, this is as far as i got, it s also not tested
https://gist.github.com/tzapu/a909c357175fd8f08ce6

pieman64, 2016-02-16:
@tzapu I notice with ESP's that many sketches with reset have delays before AND after the reset. I can understand the delay before the reset can be used to give users time to see the system is being restarted but I can't understand the delays after the reset. Generally my Wemos D1 mini responds pretty well to a reset but it is far from 100% effective. What is your understanding of the delays after reset and any particular time frame (I see 5 seconds regularly)?

tzapu, 2016-02-17:
well, in my case, some time ago, i would just have the module not restart if there was not a delay after.
similar to how deep sleep needs a delay after it to work.
the 5 seconds is arbitrary, it doesn t wait that amount of time anyway. it could possibly work just as well with 0 or 100 ms
since it s not hurting anything, i guess it s better to just leave them there and forget about it :)

closed 2018-01-26 by tablatronix