tzapu/WiFiManager #110: AP IP Came up as 192.168.244.1

pieman64, 2016-02-24:


tzapu, 2016-02-25:
no idea, have never seen that before...
for those cases the captive portal should redirect you to the proper ip as well, or actually because i reroute dns queries as well, entering any website should redirect to the proper ip...

but maybe it was printing wrong? did you actually connect to 244.1?

pieman64, 2016-02-25:
Yes I connected to 244.1 and it was fine but it wouldn't have worked without Serial Monitor telling me the required IP to use. I didn't try 4.1 but are you saying 4.1 would be ok? I think the final confirmation when I saved the parameters took me to 4.1.

pieman64, 2016-02-27:
I took my device to a friends house yesterday and I noticed when I clicked the WiFi Access Point it automatically brought up the required web page. I think this was with Firefox and I normally use Chrome and I hadn't seen this happen before. His PC is Windows 8 / 10 and my machine is Windows 8.1. Is this feature specific to certain browsers / operating systems?

tzapu, 2016-02-29:
i assume that would be the captive portal?
yeah, very os specific, works for me on os x, brings an unbranded webview, on ios, same thing

did not try on anything else

pieman64, 2016-02-29:
Not captive portal, just regular WiFiManager. Same thing happened again last night when I went to his house. I asked him which OS he has and he said Windows 8.2 which AFAIK doesn't exist so I guess 8.1.

I tried setting Firefox as my default browser and Win 8.1 but I can't replicate it on my system.

tzapu, 2016-02-29:
well, the captive portal is simply wifimanger, it s the same page really, no distinctions.
since wifimanger hijacks dns as well, maybe he had some refreshing page or something, and it redirected

kosmonavtX, 2016-06-27:
I see similiar problem of  @pieman64: 192.168.244.1 rather than the regular 192.168.4.1

tzapu, 2016-06-27:
hi guys,

sorry i don t have any more information related to this. i have only seen it happen when my flash got corrupted and i was only able to make it go away after fully erasing flash with esptool.py i think, i remember it being hard and annoying to sort out.

i know in the latest 2.3 or maybe in the github core there s been more improvements so umm_malloc does not corrupt the flash, but i supposed that if it’s corrupted, the checks won t help much. 

if this is even related to that that is…

kentaylor, 2016-06-29:
Flash corruption is a known issue, caused by a bug in the Arduino core code which will soon be fixed. 

I've had to erase the flash on a few ESP8266 devices and written an Arduino sketch to do it which is at https://github.com/kentaylor/EraseEsp8266Flash

duchere, 2016-08-10:
Hi, I am experimenting the issue. The only fix I found is to modify startConfigPortal, by adding
  WiFi.mode(WIFI_OFF);
  delay(2000);
just before
  WiFi.mode(WIFI_AP_STA);
A lower delay might work. 
When I do that, I get 100% of the time 192.168.4.1
Otherwise, I get very often 192.168.244.1
Looks like it's related to ESP8266 core

Best, 

Jean

smokeypell, 2016-08-10:
This is set in the ESP8266WiFiAP.cpp starting at line 151
`    // check IP config
    struct ip_info ip;
    if(wifi_get_ip_info(SOFTAP_IF, &ip)) {
        if(ip.ip.addr == 0x00000000) {
            // Invalid config
            DEBUG_WIFI("[AP] IP config Invalid resetting...\n");
            //192.168.244.1 , 192.168.244.1 , 255.255.255.0
            ret = softAPConfig(0x01F4A8C0, 0x01F4A8C0, 0x00FFFFFF);
            if(!ret) {
                DEBUG_WIFI("[AP] softAPConfig failed!\n");
                ret = false;
            }
        }
    } else {
        DEBUG_WIFI("[AP] wifi_get_ip_info failed!\n");
        ret = false;
    }`

I have no issue getting 192.168.4.1. My question is, how do I change what it generates for a default IP?
I know I can use setAPStaticIPConfig(), but I'd still like to know how it's generating 192.168.4.1.

tzapu, 2016-08-11:
192.168.4.1 is the SDK default, that s just what it's set to give out in lack of other settings being made.

192.168.244.1 seems to come out in case of flash data corruption. @kentaylor made a vey nice util to erase the flash completely. you should probably do that and update your esp core to the latest and hopefully you won t have that issue anymore....

thanks for all the info everyone

duchere, 2016-08-11:
@tzapu : the problem is: once in production, with a user, I can't erase the flash or do anything like this. Any idea about why setting Wifi off and delaying 500 ms before doing WiFi.mode(WIFI_AP_STA) fixes the issue ?
By the way, in addition to fixing the issue, it has tremendously improved the rate of success of connection to the AP, and request handling. 
For real, doing this hack, has multiplied the success rate, at least by 5. Weird.

eyp, 2016-08-16:
It just happened to me now, but after switch it off and on again, it started with the right IP address

pieman64, 2016-08-16:
@eyp which ESP do you have. I can't remember now if it was my ESP 07's or WeMos Mini's that came up with 244.1. Was it also the very first time you had flashed the ESP?

eyp, 2016-08-16:
Mine is ESP-12E 

No, I have flashed it lots of times.

arihantdaga, 2017-12-06:
I faced the same thing today..
Suddenly .. 
Any idea why its happening ?

closed 2018-01-26 by tablatronix

fixduino, 2018-03-20:
i have update like this..:

`boolean  WiFiManager::startConfigPortal(char const *apName, char const *apPassword) {
  WiFi.mode(WIFI_OFF); //fixduino
  delay(2000);  //fixduino
  //setup AP
  WiFi.mode(WIFI_AP_STA);
  DEBUG_WM("SET AP STA");

  _apName = apName;
  _apPassword = apPassword;`

and make  result stabil in AP IP ADDRESS 192.168.4.1.