tzapu/WiFiManager #111: signal icons

tablatronix, 2016-02-26:
I was throwing this together for myself and will probably commit to my fork and do a PR, thoughts ?
it is fairly small, slightly larger than the lock icon alone and the markup is also small, only a few classes on elements. But i do not know what is acceptable, could easily be changed to font svg icons or remove transparency to adjust size, as far as retina goes, I have no idea what the minimum size for icons should be, these might look terrible, i have not tested yet.

[image]

tzapu, 2016-02-27:
these looks awesome, really nice work
for retina, they need to be about double the size they are set at, so 20px for a 10px shown image.
about 14px seems to work ok in relation to the current font size

really excited for this, looks really nice :+1:

tablatronix, 2016-03-25:
ok so I have a final version coming.

I few things I found.

base64 encoding does not support @2x image scaling.
And I cannot just use large images and scale down for non retina, because then it is blurry, blegh

So I had to include 16px sprites, and a 32px sprite for 192ppi display using a @media selector, which I then half scale,  so the sprite coords do not have to be duplicated. It works, much crisper icons on IOS safari and chrome, I have not the hardware to test on android to see if it gets blurry or not

If you don't want the extra progmem space, you can easily comment the hi res out, no big deal.

All in all all entire style string is ~1.4k, with negligible markup added to the html.

I also added accessibility title and aria-labels to the icon container since they are background images and not very screen reader compliant by default. This also allows hover text of the quality nicely for more detail.

icons can be left or right aligned with respect to the SSID link, `left` or 'right` classes on the parent div.
There is also an 'invert' class for that to invert the icons to white. 

[image]

tablatronix, 2016-06-04:
any chance to test?

tzapu, 2016-06-05:
hi mate, sorry for taking so long, had and still have a few long and busy weeks.
i hope to make more progress next week

tablatronix, 2017-02-08:
any chance in hell of merging this?

tzapu, 2017-02-09:
hi, sure, if you could pr it to dev it would be perfect
cheers

tablatronix, 2017-02-10:
Do you have tagged issues or a milestone of the changes ofmplans for dev or a single issue even?

tablatronix, 2017-08-28:
- [ ] add template options for icon, RSSI, pecentage, might want icons + RSSI db togather, or percentage by itself. etc.

closed 2018-01-26 by tablatronix