tzapu/WiFiManager #115: AP connect failures when stored client fails.

tablatronix, 2016-02-29:
So this is a pretty normal thing. You move your board from one network to another ( I do this everyday for development )

And I noticed if there is a stored config, and it fails to connect or find the SSID, then soft AP mode does 2 things.

1. connection failures on some devices, method unknown
2. web responses are very slow or fail entirely.

I am tracking this issue as similar
https://github.com/esp8266/Arduino/issues/1094

also see my notes on it.

I added an explicit disconnect to 
`startConfigPortal()` and it seems to have stabilized my setup. ( IOS always fails to connect to AP )

I am not sure why changing mode doesn't handle this alone, but it seems some channel scan or something keeps running in the background ( no debugging revealed what it was ), maybe some interrupt problem or memory problem or by design.

tzapu, 2016-03-01:
nice investigative work:
just adding disconnect() in start config portal would have 2 very important side effects.
disconnect() erases the config, so esps that would be started and enter config due to temporarily missing router for instance, will never connect again. also people already starting config mode while connected to something, would loose connection (don t know if this is not happening already)
there may be a few things to try here, maybe you can help me with since you are getting the issue ( i am not getting it on any modules currently...)
1. add ESP.eraseConfig() at the beginning of your sketch, before anything else. does it help?
2. instead of disconnect to startConfigPortal, try this
```
WiFi.persistent(false);
WiFi.disconnect(true);
WiFi.persistent(true);
```
and maybe it won t erase the settings. this should be tested as well, that if you reboot, instead of reconfigure after starting the portal, it would still remember the old settings. i think this is very important
3. try the ap + sta mode instead of ap? maybe that s more...compatible? although in the past there s been lots of issues with that

thanks

tablatronix, 2016-03-01:
Eraseconfig requires a reboot but yeah the problem goes away cause theres nothing to connect to.

Disconnect should not erase config it should turn off radio. Maybe i read wrong cause that makes no sense.

tzapu, 2016-03-01:
if persistent is true, disconnect will also erase config. that s what i used in resetSettings, seems very effective

erase config deletes calibration data, so if it s done before connecting to anything, should it not help? ( i haven t look in what it actually does, i just saw the recommandation at some point by igrr to put it in front of everything, before connecting, etc...i think...if i remember correctly)

tablatronix, 2016-03-01:
Erase config wipes your config file so it requires a reboot to take effect and like you said you lose config which could be bad if you make mistake or cancel config

tzapu, 2016-03-01:
i tried it without reset so that s why i didn t see the erase of ssid pass as well
https://github.com/esp8266/Arduino/issues/1494

tzapu, 2016-03-01:
so yes, 1 doesn t seem to be a go...

tablatronix, 2016-03-01:
Yeah its probably async I always manually restart never tried automating it.

tzapu, 2016-03-01:
well, i looked at it, it actually wipes the flash sectors cotaning the data, so it matters when that data is read, to see if it needs a reset or not

tablatronix, 2016-03-01:
I can't even find the definition for WiFi.disconnect in the source, Did you see it even documented somewhere?

tzapu, 2016-03-01:
here 
https://github.com/esp8266/Arduino/blob/bbaed29b7e55f15d61eeedbf4d91e0473672222d/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L243

tablatronix, 2016-03-01:
yeah not sure how I was missing it.

So I cannot reproduce this outside of wifimanager. I am trying to replicate as close as possible, but no luck. Strange.

```
/* based on wifiaccess example */
/* Create a WiFi access point and provide a web server on it. */

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "ESPap";
const char *password;

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
 * connected to this access point to see it.
 */
void handleRoot() {
	server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
	delay(1000);
  Serial.begin(115200);
  Serial.setDebugOutput(true); 

  // WiFi.begin("nonexisting","password"); // store
  WiFi.begin(); // restore
  
  delay(300);
  WiFi.printDiag(Serial);

  Serial.println((String)WiFi.waitForConnectResult());
  
  WiFi.mode(WIFI_AP);
  delay(300);
  WiFi.printDiag(Serial);
  
	Serial.println();
	Serial.print("Configuring access point...");
	/* You can remove the password parameter if you want the AP to be open. */
	WiFi.softAP(ssid, password);
  delay(300);
  WiFi.printDiag(Serial);

	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);
	server.on("/", handleRoot);
	server.begin();
	Serial.println("HTTP server started");
}

void loop() {
	server.handleClient();
}
```

tzapu, 2016-03-01:
i ve been trying the same with no luck, but did not get to webserver level like you
i suggest adding the store part to the handleRoot, and a connect after that?

that should simulate the client being connected as well

tzapu, 2016-03-05:
hi, i ve noticed another scenario that seems to be leading to a deadlock in WiFi.waitForConnectResult()
it seems to be related to the auto connection(of the esp) mechanism and WiFi.begin(); 
you could try disabling the auto connect of the sdk on boot, i know there s a flag for it somewhere

i ran into it and added a possible fix to wifi manager, but that only works when you try with a good user and password

tablatronix, 2016-03-05:
Yeah its a odd thing I never did figure it out, but its a timing issue of where in an ap autoconnect failure scan process you try to start the softap, I am thinking it is something to do with a reset state or which channel the scan is on or landed on (it ends at 14, which is not supported by some hardware) so maybe that is why my IOS has issues. But I gave up for a while until I can get more advanced debugging and find out where that code is and how it works.

tzapu, 2016-03-05:
the issue in waitForConnectResult i found, seems realted to the fact that the SDK starts connecting on boot, even before we call any functions in wifimanager
need to find those fucntions to disable that

tablatronix, 2016-03-05:
I find that if you can do your stuff before it even has a chance then you are usually ok.
But if you wait or there is a delay (which waitforconnect does ) then you have to wait even more or change mode and then wait for it to stabilize, so adding a delay of about 2000 after either of those seemed to fix my problems but that could just be random timing issues also.

Either way autoconnect = false should theoretically resolve, but i guess it saves and then you need to restart for it to take effect also.

tzapu, 2016-03-08:
hi, finally i was able to reproduce this outside wifi manager
could you tell me what WiFi.status() returns for you in the example above, immediatelly before the locking waitForConnectResult() ?

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

tablatronix, 2016-03-09:
paste the commit hashes ?

It does seem more stable now.

tzapu, 2016-03-10:
fixes in these 2 hashes 
b99487785d2c8319df1223b204e1bba2b321f3f3
566bcbe0632c2c501995a7a9d6c3490bb0d33e5b

that is great news

tablatronix, 2016-03-10:
ahh, interesting fixes.

sej7278, 2016-07-10:
did anyone figure out how to do this yet, i "persisted" an accesspoint i've never even connected to and now it won't even try other AP's.

tablatronix, 2016-07-10:
Do you have alot of AP in the area ?

You have 2 options as this thread notes.

add WiFi.disconnect(true); before your ap connect in your sketch.

erase_config using esptool

sej7278, 2016-07-10:
this does absolutely nothing, doesn't even blink the serial led i don't think:

https://github.com/igrr/esptool-ck/#erase-flash-on-a-nodemcu-board

running this a few times seems to do it randomly:
```
WiFi.persistent(false);
WiFi.disconnect(true);
WiFi.persistent(true);
```

now i've got to figure out why it doesn't seem to be able to connect to g-only ap's.

tablatronix, 2016-07-10:
I never could get esptool.exe to erase flash, i had to use esptool.py

kentaylor, 2016-07-11:
I did an Arduino sketch for flash erasure which is at https://github.com/kentaylor/EraseEsp8266Flash

sej7278, 2016-07-11:
nicely documented too @kentaylor

e1red, 2016-08-10:
Hey guys,

Figured something out!  Turns out the host definition in both the WiFiClientBasic and WifiClient example scripts was wrong.

Change this:
`const char * host = "192.168.4.1"; // ip or dns`
To this:
`const IPAddress host = IPAddress(192,168,4,1);`

Something about the way the libraries define whether we are asking for a IP or a DNS.  Especially working with teeny dumb boards (I am on a NodeMCU 0.9).  And I could connect to google via my home wifi but then couldn't seem to connect to 192.168.4.1 on my second NodeBoard

And wow that's like 10 hours of my life I won't get back.  Also the WifiClientBasic script doesn't have a GET command so doesn't ever print out the webpage, but at least the WifiClient example script does.

Things that didn't help
WiFi.setPhyMode(WIFI_PHY_MODE_11G);
WiFi.disconnect();
WiFi.mode(WIFI_STA);
the erase script by KenTaylor (thanks but I guess not relevant right now)