tzapu/WiFiManager #116: If client save connect fails , no timeout

tablatronix, 2016-02-29:
If you save settings , and the client connect fails, wrong password etc, the connection never fails, you have to reboot the board to get configportal back.

Would be nice to auto handle this. Running STA+AP would be neato since you get real time status, but I think that has too many issues currently ( channel switching disconnects etc.)

tzapu, 2016-03-01:
this is a weird one, while the config portal is running, before connect, it is actually kind in STA + AP
(probably the issue of random issues people have)
the behaviour i get sometimes is that it fails connecting and i can click back on the config portal browser window to rescan, other times, portal config goes away, but only when wifi gets disconnected as well, and in this case i just tap the AP name, to reconnect, and re get the config portal..

this is on a ios device, and on os x, not sure how it works on others, what device are you using ? can you not just reconnect to wifi?

i am thinking of making a "safe mode" switch, that would change the current behaviour to connect to disable softAP, kill web and dns server, try to connect, if fails, start everything up again, for people that have issues currently with how it works at times...

kentaylor, 2016-05-02:
The version at https://github.com/kentaylor/WiFiManager/ keeps the configuration portal open until it is shut down by a button press in the portal interface. This caters for the @tablatronix scenario. 

When the configuration portal automatically closed and the device WiFi network automatically disappeared, as occurred previously,  browsing devices usually change automatically to another WiFi network. User testing showed that most people thought they were still on the device WiFi network and interpreted the inability to access the configuration portal as the device having failed. Restarting the device didn't fix it for them as being now on another WiFi network they could still not access the configuration portal.

closed 2018-01-26 by tablatronix