tzapu/WiFiManager #117: Seems there is no DHCP timeout

jlplaza, 2016-03-05:
My setup: router connecting internet, serving IP in lan and wifi repeater giving new ssid, but relying in router to serve IP. WifiManager configured and working to autoconnect with timeout to second ssid.
If I power off main router and reboot ESP, it finds second ssid, connects but as it can't get IP the system blocks forever waiting (neither failback to portal nor the failback function).
Does it mean that if DHCP server hangs all the ESP in lan will be unrecoverable in reboot? 
Thanks

tzapu, 2016-03-05:
that sounds to me the same behaviour any device like a laptop would have, if there s no dhcp, the device won t get an ip, and won t work

depending on your configuration, and where it locks, you could add a check after autoConnect, and if IP is not ok, reboot.

jlplaza, 2016-03-06:
the problem is I can't check anything after autoconnect because it never ends (autoconnect waits forever for the ip). Maybe It would be nice to timeout autoconnect after a fixed time, let execute the callback function to do offline tasking and continue to portal with timeout to loop again

tzapu, 2016-03-07:
hi, 
thanks for this, it seems there are a number of edge cases, like yours, that i am now discovering, that involve autoConnect/waitForConnectionResult to stall forever. I will need to find a way to test this myself.
in the mean time, i have added a change in the latest commit that sorts at least some of these cases, probably not yours.
There s also a function to force a timeout on connection, mentioned here https://github.com/tzapu/WiFiManager#debug

try adding `wifiManager.setConnectTimeout(60);` BEFORE autoConnect
please let me know what happens, these kind of errors have been making my life miserable, as people report some of them but I can t replicate :D

tzapu, 2016-03-08:
hi, i ve just made another commit to this, could you please try the github version of WiFiManager and see if anything changed for you?

closed 2018-01-26 by tablatronix