tzapu/WiFiManager #120: A bit more on-demand

richardwilliamson, 2016-03-06:
Hello,

Great Library!

I've put in a couple of pull requests, hope I've done it right as I haven't done a great deal with Github before. I think my use case may be a bit different to that intended so I thought I would put a couple of notes..

- I am using the library with a remote control device which has an interface (with buttons), the user can start up the web-config on demand when required
- The user should be able to decide whether to have a static or DHCP IP 
- The user may set the device up while not in range of the current network

The above I think I have made work with the pull requests I put through, there are a couple more things I still need to look at:

- The user may go into the AP for reasons other than changing the network, so I am thinking about making an option where the SSID and password can be pre-populated with the current details that are in use (even if they don't work)

Thanks!

MiloshCZ, 2016-03-24:
Hi richardwilliamson,
I agree with everything what you said :-D

tzapu, 2016-03-24:
hi, sorry for not getting around to this yet, it s been a hectic week.

i want to document and launch the current version, then add this. 

thank you very much for both pull requests and appreciation and feedback

richardwilliamson, 2016-03-24:
Hello

That's great! Thanks!! I will put in another pull request soon too as I have done some other stuff which you may or may not like!

Richard

tzapu, 2016-03-24:
bring em on :d

closed 2018-01-26 by tablatronix