tzapu/WiFiManager #122: Reconnect?

matthuisman, 2016-03-07:
Just wondering how WiFiManager manages WiFi disconnects.
Does it automatically try to reconnect etc?

matthuisman, 2016-03-07:
I just tried and it doesn't seem to.

Is there any code to make it re-connect.

I would have this in my loop function

void loop(void){
  if (WiFi.status() != WL_CONNECTED) {
    disconnects++;
    Serial.println("Disconnects: "+String(disconnects));
    wifiManager.reconnect()  ????
    return;
  }
  server.handleClient();
}

matthuisman, 2016-03-07:
OK.
Using autoconnect again will not work, as that may start the config portal instead of waiting for Wifi.

So, looks like

wifiManager.connectWifi("","");

would be the way to go.

tzapu, 2016-03-07:
hello,

Usign WiFi.begin() should be fine and work togther with wifi manager.... but... the sdk itself has some auto reconnect functions. it should do it, indifferent of WiFiManager usage, once it s connected

This is my guess, and observations of the esp8266 core and nodes i have online, i have not double checked exactly what is happening behind the scenes

tablatronix, 2016-03-09:
it is supposed to

```
bool ESP8266WiFiSTAClass::setAutoReconnect(bool autoReconnect) {
    return wifi_station_set_reconnect_policy(autoReconnect);
}
```

hemalchevli, 2016-04-05:
I'm facing a similar problem of not reconnecting after power cycle. I'm using on demand configuration with 18 custom parameters(changed` WIFI_MANAGER_MAX_PARAMS` in header).
As a work around I'll try adding ssid and password as custom parameters and use the following in loop to reconnect
`if(WiFi.status() != WL_CONNECTED){
    WiFi.begin(ssid,password);
  }`

tzapu, 2016-04-05:
hi, are you using the latest github version or a release?

hemalchevli, 2016-04-05:
I downloaded it via PlatformIO library manager, so it must be a release as far as I know.

tzapu, 2016-04-05:
hi, 
jsut released a new version, 0.11 about 1 hour ago.

don t know if it appears on platform io yet. this one has some fixes for connection issues in it

cheers
alex

hemalchevli, 2016-04-05:
I'll manually download it and copy to it to lib folder and check if the issue still persists.
thanks

Great lib by the way!
Cheers!

tzapu, 2016-04-05:
thank you very much

hemalchevli, 2016-04-05:
Okay so I did 10 power cycles, and esp connects every time, and also it does not lose connection with time. Looks like problem solved.
Thanks for the support @tzapu

tzapu, 2016-04-05:
no problem, thanks for using it and the feedback
this is great news

closed 2016-09-27 by matthuisman

Gregauriz, 2018-09-20:
I'm still having a similar issue (not sure it's the right place to put it), It reconnect fine if during the setup it is was able to connect to a wifi (or setup vi the portal have been done). But if during setup nothing is done on the portal and there is no wifi available, there is no way to do a reconnection ? I try to do an autoconnect in the loop but it wasn't able to get a connection because the wifi is in WL_IDLE_STATUS...

So here an example of code : 
`void setup(){
WiFiManager wifiManager;
wifiManager.setConfigPortalTimeout(120);
wifiManager.setTimeout(20); 
wifiManager.autoConnect('XXXXXXXX');
}
void loop() {
if (WiFi.status() != WL_CONNECTED) {
      Serial.println("Trying to connected to wifi");
      WiFiManager wifiManager;
      wifiManager.setConfigPortalTimeout(120);
      wifiManager.setTimeout(20); 
      wifiManager.autoConnect('XXXXXXXX');
    } else {
      // do stuff
    }
}`

This code give this in the serial : 
tarting WifiManager
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: 
*WM: Configuring access point... 
*WM: XXXXXXXX
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: **0**
*WM: 
*WM: Configuring access point... 
*WM: XXXXXXXX
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

I hope to clear enouth ! Feel free to correct me if I'm wrong... I tried to add a `WiFi.begin();` but didn't change anything, it's stuck in idle mode...

tablatronix, 2018-09-20:
new issue pleaswe

fzoccara, 2018-09-25:
> I'm still having a similar issue (not sure it's the right place to put it), It reconnect fine if during the setup it is was able to connect to a wifi (or setup vi the portal have been done). But if during setup nothing is done on the portal and there is no wifi available, there is no way to do a reconnection ? I try to do an autoconnect in the loop but it wasn't able to get a connection because the wifi is in WL_IDLE_STATUS...
> 
> So here an example of code :
> `void setup(){ WiFiManager wifiManager; wifiManager.setConfigPortalTimeout(120); wifiManager.setTimeout(20); wifiManager.autoConnect('XXXXXXXX'); } void loop() { if (WiFi.status() != WL_CONNECTED) { Serial.println("Trying to connected to wifi"); WiFiManager wifiManager; wifiManager.setConfigPortalTimeout(120); wifiManager.setTimeout(20); wifiManager.autoConnect('XXXXXXXX'); } else { // do stuff } }`
> 
> This code give this in the serial :
> tarting WifiManager
> *WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Using last saved values, should be faster
> *WM: Connection result:
> *WM: 1
> *WM:
> *WM: Configuring access point...
> *WM: XXXXXXXX
> *WM: AP IP address:
> *WM: 192.168.4.1
> *WM: HTTP server started
> *WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Using last saved values, should be faster
> *WM: Connection result:
> *WM: **0**
> *WM:
> *WM: Configuring access point...
> *WM: XXXXXXXX
> *WM: AP IP address:
> *WM: 192.168.4.1
> *WM: HTTP server started
> 
> I hope to clear enouth ! Feel free to correct me if I'm wrong... I tried to add a `WiFi.begin();` but didn't change anything, it's stuck in idle mode...
> And I don't want to reset the all ESP....

Hi @Gregauriz ! thank you very much! this help me a lot.. 
just one note: "setTimeout" seems to be an alias for "setConfigPortalTimeout" function.

Best.
Francesco.

tablatronix, 2018-09-25:
Its not an alias, its deprecated

rodrigok1, 2020-06-08:
dear All,
Is there any way to find the SSID and password saved by WIFImanger lib?
If so, it will be simple to reconnect in case of a Wi-Fi fall during the program execution.
Can anyone help me or US?
I appreciate!