tzapu/WiFiManager #123: Console window should use html special character escaping

krahabors, 2016-03-07:
Console window in web interface does not show serial output, which contains html-formatted strings, <html> for example. That output is present in telnet session, connected to the same device, but is completely missing in web console.

tzapu, 2016-03-07:
hello,

i am confused, which console window in web interface?

cheers
alex

krahabors, 2016-03-07:
Ooops, sorry, it's wrong project I commented in:)
Please delete this issue, if possible.

tzapu, 2016-03-07:
:))), no worries, if it s another project of mine, please let me know through an issue
if there another similar project, i d also like to know :D, i like cool projects :D

closed 2016-03-07 by tzapu

krahabors, 2016-03-07:
No, that was about jeelabs/esp-link, which I also use extensively, together with yours excellent tool.