tzapu/WiFiManager #126: Error with include order in docs

SeanBannister, 2016-03-15:
In README.md it lists 

#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

tzapu, 2016-03-15:
this is how i have it in all my sketches
```
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager
```
not working for you ?

SeanBannister, 2016-03-15:
As soon as I change it to that I get the following errors (but by switching them it works fine):

```
In file included from C:\Users\Sean\AppData\Local\Temp\arduino_15fccb0d51e8edc29ac5d81953b4c711\sketch_mar16a.ino:2:0:

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:3: error: 'WiFiClient' does not name a type

   WiFiClient client() { return _currentClient; }

   ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:127:22: error: 'WiFiClient' has not been declared

   bool _parseRequest(WiFiClient& client);

                      ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:130:19: error: 'WiFiClient' has not been declared

   bool _parseForm(WiFiClient& client, String boundary, uint32_t len);

                   ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:133:27: error: 'WiFiClient' has not been declared

   uint8_t _uploadReadByte(WiFiClient& client);

                           ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:143:3: error: 'WiFiServer' does not name a type

   WiFiServer  _server;

   ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:145:3: error: 'WiFiClient' does not name a type

   WiFiClient  _currentClient;

   ^

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h: In member function 'size_t ESP8266WebServer::streamFile(T&, const String&)':

C:\Users\Sean\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:121:10: error: '_currentClient' was not declared in this scope

   return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);

          ^

exit status 1
Error compiling.
```

tzapu, 2016-03-15:
what arduino ide are you using?

SeanBannister, 2016-03-15:
I was running Arduino IDE 1.6.7 so I upgraded to 1.6.8 but I'm still experiencing the same issue.

ESP8266 core for Arduino - 2.1.0
WifiManager - 0.10
Windows 8.1 Pro

tzapu, 2016-03-15:
hi, do you include ESP8266WiFi.h after those 3 includes when it errors?

SeanBannister, 2016-03-15:
No, but here's the simplest example that causes the error:

```
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

void setup() {
  WiFiManager wifiManager; 
  wifiManager.autoConnect("test", "12345678");
}
```

tzapu, 2016-03-15:
try adding this on top of all those

```
#include <ESP8266WiFi.h>
```

SeanBannister, 2016-03-15:
That works :) thanks for your help.

Maybe this could be added to the readme.md?

tzapu, 2016-03-15:
ok, thanks
i should probably add it then if it s not clear

cheers

tzapu, 2016-03-16:
i ve added this now, thanks

closed 2016-03-16 by tzapu