tzapu/WiFiManager #127: Webserver Hostname Configuration for DHCP Headless Setup

Grimace06, 2016-03-17:
Hey All-

Love this utility. Working on my 1st webserver project and I'm trying to build a unit that can be set up headlessly by an inexperienced user (sister & brother in law).

The WiFi manager sets up the webserver perfectly and I'm able to set the hostname by using the command [wifi_station_set_hostname("foobar");], but I'm unable to access the webserver by hostname, only by IP address. As the end user will not be able to see the IP address, is there any way to set up a static web browser navigable feature so that the user can find the server pages without knowing the DHCP IP?

I haven't yet used the Angry IP scanner, as I hear that works well to detect connected local peripherals, because I'd like to make this as easy as possible for my non-tech inclined sister & bro inlaw.

Thanks for your insight!

-KDG

tzapu, 2016-03-24:
hi, thank you for the appreciation
is this hostname necessary for dhcp ? 
or just regular hostname that you type in browser to open page? 
when the portal is started, and you are connected to it, it also starts a dns server that should reroute any domain to the config portal... can you check?

tzapu, 2016-05-08:
Hi,
wifi_station_set_hostname or WiFi.hostname set the DHCP hostname it seems.
For accessing through a webserver, any hostname should work as the dns captive portal should redirect everything to it.
If there is a problem with that, please reopen the issue.

Thank you very much for using it, love the appreciation :D

closed 2016-05-08 by tzapu