tzapu/WiFiManager #130: Add password to AP with generated name

javl, 2016-03-20:
I tried to expand the code so you could have the ESP generate its own AP name, but also provide a password. I tried doing this by adding a `wifiManager.ssidToName()` function that returned the generated AP name, but something goes wrong when passing it to `autoConnect()`. I thought it worked in my pull request #128 but it showed up as a bunch of weird characters in the Wifi menu on my computer.
Any idea if this would be easy to implement?

Using this, you could easily set up a large amount of devices (because of their unique AP names) but still have them somewhat protected while you work your way through setting them all up.

tzapu, 2016-03-21:
hi, auto generation is already in, without password though.

i ll see about adding auto generated name, and password, need to rework some of that as well

javl, 2016-03-21:
I saw the autogenerate was already there, but I tried to add a password to it. But I wasn't sure how to pass the generated AP name to the autoConnect() function (it seems to expects a pointer as the apName argument?)

ozayturay, 2016-08-20:
```
String APName = "ESP" + String(ESP.getChipId());
wifiManager.autoConnect(APName.c_str(), "DEADBEEF");
```
works but

```
String APName = "ESP" + String(ESP.getChipId());
const char* SSID = APName.c_str();
wifiManager.autoConnect(SSID, "DEADBEEF");
```
does not work. Somehow the .c_str() suffix needs to be inside the autoConnect parameters not outside. 

Maybe this helps to add this feature but it is not very much needed as it just adds one more line to user code.