tzapu/WiFiManager #131: WiFiManager port to Energia/CC3200

freber, 2016-03-21:
Hi,

I'm considering porting this very nice library to Energia and wanted to check if there has been any work done already before starting.

tzapu, 2016-03-22:
hi, no, there hasn t, i do not have such a device to test with 
feel free to do so
cheers

freber, 2016-03-27:
I've looked into it now and since you are using some of the esp arduino core libraries such as dns and we server I'm considering to include these (also needs some porting) in a fork of wifi manager. Energia doesn't have the web server and DNS library.

It would however be great to be able to get my port merged back to you initial great repository but I'm afraid it would look a bit messy.

What would yo prefer/do?

tzapu, 2016-03-29:
i ve just recently seen a project that tries to cater for multiple platforms and... it s just a mess, due to restrictions like libs not being available, slight change of various functions, etc... a large, hard to understand and hard to manage mess...

my first and only thought is that this would only work without the bloat if it s kept as separate as possible, and only included via ifdef platform, to cater for the missing interfaces. does that make sense? is it even possible? 

DNS server is not really mandatory anyway, so that can be made optional, if you port the web server lib, then that would have the same interface pretty much. the wifi lib interface seems pretty similar, but there are a lot of non standard fixes in the code just to cater to the way the esps are behaving, who knows if those will work ok with the cc3200

closed 2018-01-26 by tablatronix