tzapu/WiFiManager #132: No/Gibberish serial output on Olimex MOD-WIFI-ESP8266

codmpm, 2016-03-21:
Hi tzapu,

just found WiFiManager and it is exactly what I need - great work! Sadly I ran into problems when using the [MOD-WIFI-ESP8266](https://www.olimex.com/Products/IoT/MOD-WIFI-ESP8266/open-source-hardware) from Olimex. I can't get serial output to work. All I see is something like this: "�n�����

tzapu, 2016-03-22:
hi, that looks like the boot loader support at 74880.
and then nothing from your sketch.
do you have serial begin in your sketch before wifi manager? does this happen to any example provided? how about some esp8266 examples?

codmpm, 2016-03-22:
Hi tzapu,

thanks for your fast reply. I'm using the [AutoConnectWithFeedbackLED](https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithFeedbackLED) example without any modifications. 

    void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    //...
    }

The exacty same code works on the Huzza-board. The only difference is the board definition: (http://arduino.esp8266.com/stable/package_esp8266com_index.json), as I use the appropriate ones for the Huzza- or the Olimex-board. 

Thnx,
Patrik

tzapu, 2016-05-08:
hi,

i don t have an olimex board to test unfortunately, i am not sure what could be different.

if you use generic board with appropiate settings, does it still happen?

closed 2018-01-26 by tablatronix