tzapu/WiFiManager #133: Option for separate labels on Custom Parameters

Humancell, 2016-03-23:
We are now adding a pretty good list of custom parameters on the configuration page.  They are working well, but now our users are getting confused about which field is which when they see 8 of them listed with default parameters.  Several, in our case, are host names.

Users are now forced to clear the default, or previously set value, in order to see the tip on what goes in which field.

Is there a way to add an option to put a text label outside and above the input field?

Configuration Host
[                                                    ]
Configuration Host Port
[                                                    ]

This way they can tell what is what.

I know a lot of people might not want this, but in our case it's becoming important.

Thoughts?

tzapu, 2016-03-23:
hi, in the latest github version you can add custom html to that part so you can add descriptions
use like this
```
WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "iot.eclipse", 40, " readonly");
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", NULL, 5);
  WiFiManagerParameter custom_text("<p>This is jsut a text paragraph</p>");
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", "sdfsfasdfdsfsdfa", 32);

  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_text);
  wifiManager.addParameter(&custom_blynk_token);
```

Humancell, 2016-03-23:
Oh, this is great!  So if there is only one parameter it's assumed to be a text string of HTML?

Are the paragraph tags required?  Or are there limits on what HTML can be placed in the quotes?

tzapu, 2016-03-23:
no limits, it gets inserted into the main html. it should be valid html

have fun

Humancell, 2016-04-29:
I know I'm doing something stupid here ... but I can't get this to work:

      const char* wovynFirmwareVersion     = "Firmware-1.3";

      char versionBuffer[32];    // this needs to be large enough for the version text and tags
      strcpy(versionBuffer, "<p>");
      strcat(versionBuffer, wovynFirmwareVersion);
      strcat(versionBuffer, "</p>");

      // The extra parameters to be configured (can be either global or just in the setup)
      // After connecting, parameter.getValue() will get you the configured value
      // id/name placeholder/prompt default length
      WiFiManagerParameter customVersionText(versionBuffer);
FAILS HERE  ^^^^^^^^^^^^^^ no matching function for call to 'WiFiManagerParameter::WiFiManagerParameter(char [32])'

      //set config save notify callback
      wifiManager.setSaveConfigCallback(saveConfigCallback);

      //add all your parameters here
      wifiManager.addParameter(&customVersionText);

What am I doing wrong?

tzapu, 2016-05-08:
hi, did you update to the latest WifiManager version?

cheers

Humancell, 2016-05-08:
I was using the "Library Manager" to install/manage, and it says I'm using v0.9.0 ... and that is beyond what can be selected in the version drop down ... so I assumed I am.

What is the current version?

Humancell, 2016-05-08:
I just saw the readme update that says v0.12 is the latest.  Any idea why this is not available or working via the Library Manager?

I'll download from Git, but have to now see how to merge that with the Library Manager installed version ... not sure what I have to do to switch install versions.

tzapu, 2016-05-09:
hi, you jsut need to overwrite the lib in it s folder with the github one.

i think there were some issues on windows with the library manager caching version and not offering updates...

Humancell, 2017-01-05:
This is working great!

closed 2017-01-05 by Humancell