tzapu/WiFiManager #135: setting up a static IP for the normal usage (when not in STA mode)

Jobenas, 2016-03-24:
Hi tzapu!

First of all, congratulations on the library is really awesome and a life saver! It actually motivated me to take up esp projects again. I wanted to know if there's a way to tell the Wifimanager to try to connect to a network with a given static ip? I've been reading the documentation but I can't seem to figure how to do that in particular. Thanks for any help on that regard.

Jorge.

tzapu, 2016-03-24:
hi,

this is what you need

https://github.com/tzapu/WiFiManager#custom-station-client-static-ip-configuration

cheers and thank you very much :D

closed 2016-03-24 by tzapu

Jobenas, 2016-03-24:
From what I read that gives the STA, the station portal, a static IP. Am I understanding correctly that this wouldn't affect the application once it connects to a network? Thanks again!

tzapu, 2016-03-24:
hi

STA = station (client)
AP = access point (configuration mode)

so using that it would set the IP after it connects to the network

cheers

Jobenas, 2016-03-25:
Thank you very much! didn't get that definition