tzapu/WiFiManager #137: Save Parameter with startConfigPortal

WilliamFrasson, 2016-03-25:
Hello
I find a problem tring to use the startConfigPortal adding parameters. Maybe i'm doing something wrong..

1)  after running the startConfigPortal and press save button,  using the callback I try to readback the parameter  using the ".getValue()" se below the code. But the strings are empty..
2) more over the Setting of SSID and Password are empty (seeing the WM log, as you can see below)
3) last issue:  Ater closing the configuration portal on demand.  How i can rea SSID and PSW for start the wifi connection (I woudn't use the Auto connection because of i don't want sart the configuration portal ..but (i not so expert in this beautiful library)  how i can start the the connection using the ssid and password saved after the use of the portal on demand?

I'm using Arduino ide 1.6.8
esp8266 library 2.1,0
WiFiManager  0.10

code
  WiFiManagerParameter custom_http_Cloud_Address("Http Cloud address", "Cloud Adress", g_stConfigDevice.cloud_address, sizeof(g_stConfigDevice.cloud_address));
  WiFiManagerParameter custom_http_Cloud_Port("Http Cloud port", "Cloud Port", g_stConfigDevice.cloud_port, sizeof(g_stConfigDevice.cloud_port));
  WiFiManagerParameter custom_Sensor_Temp("Temp Sensor", "Temp APi Sensor Api sign", g_stConfigDevice.sensor_apikey_Temp, sizeof(g_stConfigDevice.sensor_apikey_Temp));
  WiFiManagerParameter custom_Sensor_Hum("Hum Sensor",  "Hum Api Sign", g_stConfigDevice.sensor_apikey_Hum, sizeof(g_stConfigDevice.sensor_apikey_Hum));

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  wifiManager.addParameter(&custom_http_Cloud_Address);
  wifiManager.addParameter(&custom_http_Cloud_Port);
  wifiManager.addParameter(&custom_Sensor_Temp);
  wifiManager.addParameter(&custom_Sensor_Hum);
  b_saveConfigCallback = false;

  //reset settings - for testing
  //wifiManager.resetSettings();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  wifiManager.setTimeout(120);
  //wifiManager.setConfigPortalTimeout(180);

  //it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration

  if (!wifiManager.startConfigPortal("AppDHT_AccessPoint"))
  {
    DEBUG_PRINTLN("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  Serial.println(wifiManager.getConfigPortalSSID());
  

  if (b_saveConfigCallback)
  {
    bool ret_code;
   

    //read updated parameters
    strcpy(g_stConfigDevice.cloud_address, custom_http_Cloud_Address.getValue());
    strcpy(g_stConfigDevice.cloud_port, custom_http_Cloud_Port.getValue());
    strcpy(g_stConfigDevice.sensor_apikey_Temp, custom_Sensor_Temp.getValue());
    strcpy(g_stConfigDevice.sensor_apikey_Hum, custom_Sensor_Hum.getValue());
...

LOG:

*WM: Http Cloud address
*WM: Adding parameter
*WM: Http Cloud port
*WM: Adding parameter
*WM: Temp Sensor
*WM: Adding parameter
*WM: Hum Sensor
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: AppDHT_AccessPoint
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: 204 No Response
*WM: 204 No Response
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Info
*WM: Sent info page
*WM: Info
*WM: Sent info page
*WM: Scan done
*WM: OneMorePeace
*WM: -44
*WM: Vodafone-WiFi
*WM: -91
*WM: Vodafone-30482066
*WM: -93
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: Http Cloud address
*WM: 
*WM: Parameter
*WM: Http Cloud port
*WM: 
*WM: Parameter
*WM: Temp Sensor
*WM: 
*WM: Parameter
*WM: Hum Sensor
*WM: 
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
failed to connect and hit timeout

thanks in advance

William

tzapu, 2016-03-25:
hi, 
i think that if it fails to connect, it does not call the save parameters callback. so you would need to remove that check. `if (b_saveConfigCallback)`
I can t see where from you are printing the wifi ssid and pass, but it does not look like it s connecting anyway, so something wrong there
to connect once it connected sucessfuly once, after config, on next boots you could just do
`WiFi.begin();`

i hope this helps
thanks

WilliamFrasson, 2016-03-25:
Hello, 
thanks for the hints.  
Now i call  WiFi.begin() instead of the WiFi.begin( ssid, password) . but i still see the problem of  read-back the value of parameters added:

here the last log:

Connecting to AP................
*WM: Adding parameter
*WM: Http Cloud address
*WM: Adding parameter
*WM: Http Cloud port
*WM: Adding parameter
*WM: Temp Sensor
*WM: Adding parameter
*WM: Hum Sensor
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: AppDHT_AccessPoint
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: 204 No Response
*WM: Handle root
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: Http Cloud address
*WM: 
*WM: Parameter
*WM: Http Cloud port
*WM: 
*WM: Parameter
*WM: Temp Sensor
*WM: 
*WM: Parameter
*WM: Hum Sensor
*WM: 
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: WiFi save
*WM: Parameter
*WM: Http Cloud address
*WM: 
*WM: Parameter
*WM: Http Cloud port
*WM: 
*WM: Parameter
*WM: Temp Sensor
*WM: 
*WM: Parameter
*WM: Hum Sensor
*WM: 
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Save New Configuration...
Exit from Configuration Portal.

cloud_address: 
cloud_port: 
sensor_apikey_Temp: 
sensor_apikey_Hum: 
Start Saving New Configuration... OK!

Restart Module!

So it seem to don't readback the parameter

second) how i can print ssid and password.? i check in WiFiManager.h but i don't see any method .. 

thanks in advanced

WilliamFrasson, 2016-03-26:
hello,
i found the error that was giving the problems over the .getValue when i added a my parameter (See below)

I use a ID composed of more than one world when added parameter, 
for example : using id = ""Cloud Address", give the error documented above, while using "Cloud_Address" the  .getValue return the expected value

  // id/name placeholder/prompt default length --> 
   WiFiManagerParameter custom_http_Cloud_Address("Cloud_Address", "Cloud Adress", g_stConfigDevice.cloud_address, sizeof(g_stConfigDevice.cloud_address));

now i'm going to resolve the second problem: after a reset (por, or ESP.restart();) the device don't find the SSID and PASSWORD stored before ...

maybe some one have some idea about it?

tzapu, 2016-03-26:
hi, you can use the standard 
```
WiFi.SSID();
WiFi.psk();
```
to retreive whatever the SDK stored.

I would also encourage you to update to the latest github version of the lib, there s some fixes included for connection logic

WilliamFrasson, 2016-03-27:
Thaks for the hint (in the while i found it and use. :P)   

Finally I'm now able to understand the what cause of loosing the SSID and PASSWORD configured using WiFiManager ;

I was calling the "WiFi.disconnect();" for save power by swithing off the radio side of esp chip.
but this cause to loose permanently the wifi configuration stored  by using the WiFiManager support..
I just check it using the printDialog metod:

WiFi.printDiag(Serial);
WiFi.disconnect();
WiFi.printDiag(Serial);

For my point of view WiFiDisconnect should not lost permanently the parameters but this is what i see also after a power on or a reset

From your point of view  this behaviour is what one should expect of is some not so normal ..

here a log:
1. Print Diagnostic WiFi
Mode: STA+AP
PHY mode: N
Channel: 11
AP id: 0
Status: 5
Auto connect: 1
SSID (9): NETGEAR22
Passphrase (16): yellowcartoon371
BSSID set: 0
WiFi status: 3
MAC: F1:B1:11:7F:CF:5C
SSID: NETGEAR22
Password: yellowcartoon371
2. Disconnect and wait 5 sec
3. Print Diagnostic WiFi
Mode: STA+AP
PHY mode: N
Channel: 11
AP id: 0
Status: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0

thanks in advanced for any comments and suggestion;

tzapu, 2016-03-28:
hi, WiFi.disconnect() erasing parameters is something that has been established by the esp8266 core for arduino. I agree with you that it should not erase settings, there should be a separate function for that, but i guess they had their reasons...

feel free to open a issue with them, link it here...

closed 2018-01-26 by tablatronix