tzapu/WiFiManager #139: using WiFiManager trunk  causes the compile error: 'isnan' was not declared in this scope

WilliamFrasson, 2016-03-27:
hello
using WiFiManager trunk  causes the compile error: "'isnan' was not declared in this scope"
while using the tag 0.10 this problem doesn't happen

I tring to use the trunk to improve some issue about connection but this new question is raised now.

tzapu, 2016-03-29:
hmm, are you using isnan in your code somewhere?

WilliamFrasson, 2016-03-29:
Yes.
Using 0.10 the code compile while using the truck  the compile fail to isnan
After Commenting isnan(..) calls the code compile.
..real Strange to me. But i see that there are some new #undef .. max and
min.. maybe these cause this behaviour.

tzapu, 2016-03-29:
well, they are connected, not themselves, but the library that now gets included... pain in the arse ...

i ll get back to you

tzapu, 2016-03-29:
could you try it as std::isnan(...)

cheers

WilliamFrasson, 2016-03-29:
I used standard arduino

float pippo;
...
if isnan(pippo)
..

tzapu, 2016-03-29:
no , i meant actually type it code like

`if(std::isnan(pippo))`

acca84, 2016-04-05:
Hello
Same problem here for me.
if(std::isnan(pippo)) works with wifimanager v0.11
if(isnan(pippo)) works with wifimanager v0.9

the problem is with librarys using isnan

could you do something please ?

tzapu, 2016-04-05:
so there would be a conflict in libraries not using the namespace then ? :(

that s ... not good

acca84, 2016-04-05:
Yes there is a problem with many libs, ie : thingspeak and DHT at least

ostersc, 2016-04-14:
I have this problem with released version .11 and using the thingspeak library as well.  Is there a work around?

tzapu, 2016-04-14:
yes, i ll switch back to a loop sort and rerelease

i m just away currently to sort it immediately 

thanks for letting me know

ostersc, 2016-04-14:
Some details for you of what I see with v.11 and not v.10
error: 'isnan' was not declared in this scope
error: 'isinf' was not declared in this scope

Arduino: 1.6.6 (Mac OS X), Board: "SparkFun ESP8266 Thing, 80 MHz, Serial, 115200"

Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h: In member function 'int ThingSpeakClass::writeFields(long unsigned int, const char*)':
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:778:30: error: 'isnan' was not declared in this scope
   if(!isnan(nextWriteLatitude))
                              ^
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:778:30: note: suggested alternative:
In file included from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:38:0,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22,

Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/cmath:632:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^

Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:789:37: error: 'isnan' was not declared in this scope
   if(!isnan(this->nextWriteLongitude))
                                     ^
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:789:37: note: suggested alternative:
In file included from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:38:0,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22,

Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/cmath:632:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^

Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:801:37: error: 'isnan' was not declared in this scope
   if(!isnan(this->nextWriteElevation))
                                     ^
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:801:37: note: suggested alternative:
In file included from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:38:0,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22,

Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/cmath:632:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^

Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h: In member function 'int ThingSpeakClass::convertFloatToChar(float, char*)':
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:1469:22: error: 'isinf' was not declared in this scope
   if(0 == isinf(value) && (value > 999999000000 || value < -999999000000))
                      ^
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:1469:22: note: suggested alternative:
In file included from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:38:0,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22,

Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/cmath:614:5: note:   'std::isinf'
     isinf(_Tp __x)
     ^

Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h: In member function 'float ThingSpeakClass::convertStringToFloat(String)':
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:1488:23: error: 'isinf' was not declared in this scope
   if(1 == isinf(result) && *value.c_str() == '-')
                       ^
Documents/Arduino/libraries/ThingSpeak/src/ThingSpeak.h:1488:23: note: suggested alternative:
In file included from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/random:38:0,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from Documents/Arduino/libraries/WiFiManager/WiFiManager.h:22,

Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/cmath:614:5: note:   'std::isinf'
     isinf(_Tp __x)
     ^

astrohome, 2016-04-26:
The workaround I found is:

`#include <math.h>`

.....

`if (std::isnan(value))`

tzapu, 2016-05-08:
guys, thanks for reporting this and sorry it took so long to get back to you.

i ve just pushed some changes that should sort this in the latest git

this will make it s way into the next release asap, so i ll close this issue, please reopen if still a problem

closed 2016-05-08 by tzapu