tzapu/WiFiManager #142: how to erase previous credentials?

hixfield, 2016-03-28:
I have a button on my ESP, that is (via an interrupt service routine) ment to restore factory settings.
How can I clear the existing WIFI credentials, so that after reset WIFIManager "sees" no credentials and starts-up the AP for setting them?

*failed* 
I found a reference to ESP.eraseConfig here https://github.com/esp8266/Arduino/issues/1494 to  but that just completely blocks the ESP and I needed to re-flash it! So I guess that is not the way to go...

hixfield, 2016-03-28:
I think I figured out a solution.
When the button is pressed I set a flag, and in the loop the flag is checked and then just a simple WiFi.disconnect() seams to push the WIFIManager in AP mode.
By the way, I found both the ESP.reset and ESP.restart to be unreliable, so I connected GPIO16 to the ESP RST pin, and using "software" I do a hardware reset like this. Yes... I am working with a flag, because the ESP seams to crash when I put the WiFi.disconnect() in the ISR itself...

This is part of my code:

```
void resetToFactoryDefaults() {
  WiFi.disconnect();
  delay(3000);
  hardwareReset();
}

void hardwareReset() {
  pinMode(g_nPinReset, OUTPUT);
  digitalWrite(g_nPinReset, false);
  delay(3000);
}

void isrResetToFactoryDefaults(void) {
  Serial.println("Resetting to factory defaults");
  g_bShouldResetToFactorySettings = true;
}

void setupWifi() {
  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //exit after config instead of connecting
  wifiManager.setBreakAfterConfig(true);
  //tries to connect to last known settings if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP" with password "password" and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect()) {
    Serial.println("failed to connect, we should reset as see if it connects");
    hardwareReset();
  }
...
}

void loop(void) {
  if (g_bShouldResetToFactorySettings) resetToFactoryDefaults();
}
```

tzapu, 2016-03-28:
hi,

as a general rule of thumb, don t do anything in interrupt callbacks except setting a flag, and process it in the loop. asking for trouble otherwise

WiFi.disconnect() will erase ssid/password

i use ESP.reset() sucessfully in a lot of projects, the key may be to add a delay(1000) after it. (also, not in a callback)

i hope this helps, cheers

baruch, 2016-04-06:
This ticket can probably be closed, maybe document it somewhere in the readme or provide an example.

barsznica, 2018-01-02:
Only WiFi.disconnect(); worked for me!

Worked [here](http://www.esp8266.com/viewtopic.php?f=32&t=8204&start=4) too!

trungkiendt9, 2018-03-16:
With button reset. It's done.      
if (reset_level >500) {
                WiFi.disconnect(true);
                delay(2000);
                ESP.reset();
        }

electron1979, 2018-12-05:
@yo2lts, I think something like what follows may work in Arduino IDE:

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
void setup()
{
  WiFi.disconnect();
//  WiFiManager.resetSettings();
//  ESP.eraseConfig();
//  ESP.reset();
//  ESP.restart();
}

void loop()
{
  yield();
}

tablatronix, 2018-12-05:
For factory reset you need to use esptool to erase flash also the arduino ide now has erase capability

yo2lts, 2018-12-05:
Multumesc, cred ca am reusit.
What is yield() ? 
Thank You! Work!

a-c-sreedhar-reddy, 2019-02-22:
WiFi.disconnect(false,true);

This erases existing AP credentials.

angusgrantBC, 2019-10-23:
None of that stuff worked. I searched for so long across so many resources.

I ended up digging around in the ESP code base and happened upon this little nugget

ESP_ERROR_CHECK(nvs_flash_erase());
nvs_flash_init();

This seems to have cleared my "AUTH_EXPIRE" credentials and got my WiFi working again. One caution is that it clears everything in flash. So if you are using preferences library, etc, it all gets wiped.

tablatronix, 2019-10-23:
You should not have to nvs erase to clear credentials, your memory was probably corrupt or changed from an upgrade.

angusgrantBC, 2019-10-23:
Well after a month and delayed shipping of product, I'll take any solution.  :-|

The strange thing is that values I was saving with the preferences library would save into flash, and be re-loaded after next reboot. It was the WiFi which would simply not connect always giving AUTH_EXPIRE exception. So I *think* it was related to a WiFi problem and nor corrupt memory. But maybe the corruption was just with the WiFi details that were saved.

tablatronix, 2019-10-23:
Yes if your partition is corrupt your wifi gets corrupted as things write into its memory space, so it might seem to work but some part of it is broken and messed up connections, I wish there was some kind of parity check here, also there was no protection for wifi struct schema when it was changed, I think there is now so upgrades wont break it as much. I always do a full erase if moving to a new version of sdk or lib.

angusgrantBC, 2019-10-23:
Your info is interesting and something I'll remember when doing sdk or library upgrades..

But I changed nothing. One day it was working, the next day it wasn't...
Oh well, problem worked around/solved.

Thanks for your comments!

qlalfdu, 2019-12-17:
WiFi.mode(WIFI_STA);
WiFi.disconnect(true,true);

rahulujjnkr, 2020-02-20:
my simple logic is , if we insert the false ssid and password into the program, and we also reset the esp within a program. problem may be resolved. i had try it and working fine. 
i use two pins D2 and D3 for this to reset the esp.
sorry for my English.... !!

/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/
this is the whole code.... 
/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/-/

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
void SoftReset(void);
void espreset(void);
const int BUTTON = 4;
const int LED = 0;
int BUTTONstate = 0;
// Replace these with your WiFi network settings
const char* ssid = "faslsessid"; //replace this with your WiFi network name
const char* password = "falsepass"; //replace this with your WiFi network password

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
  pinMode(LED, OUTPUT);
  pinMode (BUTTON, INPUT);
    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    wifiManager.setAPStaticIPConfig(IPAddress(192,168,33,1), IPAddress (192,168,33,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("WinLix Infotech AP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    BUTTONstate = digitalRead(BUTTON); 
  if (BUTTONstate == HIGH)
    {
      digitalWrite(LED, HIGH);
      SoftReset();
      espreset();
    }
    else
    {
      digitalWrite(LED, LOW);
    }
    
}

void SoftReset() 
{
  delay(5000);
  Serial.begin(115200);
 
  WiFi.begin(ssid, password);

  Serial.println();
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  }
  void espreset()
  {
    ESP.restart();
    }
  

this is my first ever try with arduino programming.

tablatronix, 2020-02-20:
Lol it says it right there in the code

//reset saved settings
//wifiManager.resetSettings();

lloydrichards, 2020-03-09:
I'm having a similar problem.  wifiManager.resetSettings() does not reset the credentials..  I'm literally running the base example:

`#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>          //https://github.com/esp8266/Arduino
#endif

//needed for library
#include <DNSServer.h>
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WebServer.h>
#endif
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}`

And its still giving me this in console:

`*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.178
connected...yeey :)
`

Either something is wrong with the function or its still storing the creds somewhere else?

tablatronix, 2020-03-10:
esp8266?

lloydrichards, 2020-03-10:
on ESP32 Dev Module.  The serial logs are at the bottom there:

*WM: Connection result: 
*WM: 3 
*WM: IP Address: 
*WM: 192.168.1.178 
connected...yeey :)

My workaround currently is to set up a manual config for the AP and attach to a button, but this means having to connect everytime which is frustrating

shazman-visuals, 2020-03-11:
Having same issue as lloydrichards here, running on a ESP32 Doit-devkit-v1.
_wifiManager.resetSettings()_ does not do what it is supposed to do (to reset wifi credentials).

Should mention that I'm using https://github.com/zhouhan0126/WIFIMANAGER-ESP32 library, because to my understanding the original Tzapu library cannot run on ESP32?

Daemach, 2020-03-11:
I run it on ESP32

tablatronix, 2020-03-12:
Development branch does

shazman-visuals, 2020-03-12:
Cool, so is there a way to reset Wifi credentials or not?
_wifiManager.resetSettings()_ is not doing the job.

tablatronix, 2020-03-12:
I have not had time to look into it, could be a bug, try erase flash also

lloydrichards, 2020-03-12:
If it helps for debugging, I was finding that even when I flashed a new program and the went back to something including the wifimanager it would still have my credentials saved when using wifiManager.autoConnect(). This doesn't seem to affect wifiManager.startConfigPortal(), but i'm guessing that cause its just overwriting the credentials.  Might give some experimenting to calling wifiManager.autoConnect() to start and then wifiManager.startConfigPortal() when i need to override

tablatronix, 2020-03-12:
the ESP saves your credentials not WM

tablatronix, 2020-03-13:
Works for me

```PHP
*WM: [1] SETTINGS ERASED 
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 Starting
*WM: [1] getCoreVersion():          2_5_2
*WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             49088
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
YES
SSID: 
PASS: 
```

tablatronix, 2020-03-13:
Let me try esp32

tablatronix, 2020-03-13:
Also works

```PHP

*WM: [1] resetSettings 
*WM: [3] WiFi station enable 
*WM: [1] SETTINGS ERASED 
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c

 Starting
*WM: [1] Free heap:        210252
*WM: [1] ESP-IDF version:  v3.2.2-132-g7dd492319-dirty
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
YES
SSID: 
PASS: 
```

tablatronix, 2020-03-13:
I cannot reproduce, I need you to erase flash, and try again and I will need more information

rodrigok1, 2020-06-03:
How to get the saved SSID and password?

peperoca116, 2020-11-20:
To get the credentials you need the following:

WiFi.begin(); // Mandatory 
delay(1000);
String TEMP_Ssid = WiFi.SSID();
String TEMP_Pass = WiFi.psk();

Done! You get credentials without the need to store them neither in spiffs nor in EEPROM

closed 2020-11-21 by tablatronix

neklauss, 2021-03-26:
The libraries built in WifiManager::erase() method did the trick for me.

CplSyx, 2023-02-09:
> For factory reset you need to use esptool to erase flash also the arduino ide now has erase capability

wifiManager.resetSettings() was not working for me (ESP8266) as autoConnect would recover the credentials - however erasing the flash via the IDE has solved the issue.

tablatronix, 2023-02-09:
Interesting, it should erase them, but maybe there was an esp bug in some version you were using