tzapu/WiFiManager #147: Memory Requirement for WiFI Manager

pieman64, 2016-04-02:
Hi @tzapu,

Having previously used WiFi Manager with 4M ESP's (ESP12) we recently moved to 1M ESP's (ESP7). To provide maximum program memory we have been setting SPIFFS at 64K but wondered if there is a minimum size required. We only have 5 parameters and WiFi Manager works on bootup but after deep sleep it stops at the following position:

```
Mounting FS...
Mounted file system
Reading config file
Config file is open
*WM: Adding parameter
*WM: blynk
*WM: Adding parameter
*WM: Firmware
*WM: Adding parameter
*WM: Max
*WM: Adding parameter
*WM: Min
*WM: Adding parameter
*WM: Calibration
*WM: Adding parameter
*WM: Timezone
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
```

If we manually reboot the ESP07 then WiFI Manager works as expected. During our testing we did find, as previously mentioned, that the IP of the AP is given as 192.168.244.1 rather than the regular 192.168.4.1. We checked access to .4.1 and it wasn't available but 244.1 worked ok.

```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: XXXXXXXX
*WM: ACCESSXXXX
*WM: AP IP address: 
*WM: 192.168.244.1
*WM: HTTP server started

```

It may be the way we are resetting the WiFi Manager settings and formatting the memory for SPIFFS as we are still not entirely sure how this is done. For example we format the SPIFFS and reset the WiFi Manager details with the two lines in your sketch but when the ESP reboots (after deepsleep or manual reset) it is trying to format SPIFFS and reset the settings again. Could you perhaps indicate the correct procedure for us to use when we want to clear the settings and start afresh.

tzapu, 2016-04-04:
hi, not sure how this will help you, but i ve only had 192.168.244.1 issues once, on one esp's flash being corrupted. stuff randomly misbehaved for a while, then it all went tits up...
i could only get it to work properly, after erasing the flash completly with esptool directly...
it was a very annoying exercise.

i do not know of any limitations of spiffs, (does not mean there aren t), 64k should really be ok...
i am using modules with deep sleep and spiffs and wifimanger, and all is ok

kentaylor, 2016-05-04:
In the version of WiFi manager you are using it calls WiFi.begin() . There is no need to do that and I think it sometimes causes failures. I suspect whether it fails depends on where is it is up to in the connection sequence. Your change of code will have changed the timing so perhaps the problem is due to the change in how long after startup WiFi.begin() is called rather than the code itself.

Try the version of WiFiManager at https://github.com/kentaylor/ and see if it helps. I'm curious as it will tend to confirm or cast doubt on my theory that calling WiFi.begin() can be problematical.

tzapu, 2016-05-04:
i am sure wifi.begin is problematic, given various circumstances.
that s why so many workarounds have been put into place :|

closed 2018-01-26 by tablatronix