tzapu/WiFiManager #148: Serving on port 80

piertoni, 2016-04-05:
Hi,
Is possible to serve pages on Port 80? Once connected the configuration page is not useful and would be great to add pages to the webpage. I managed to add another webserver on another port, but the 80 is blocked. Maybe simply transform the pointer to server from private to public?

tzapu, 2016-04-06:
hi, see examples, it shows you how to create another webserver on port 80
there s a problem with http server not deallocating as it should, or not starting up on the proper interface (you are going from sta to ap to sta)

kentaylor, 2016-04-22:
I also found that the web server could not be started more than once, not sure why. 

A workaround is to reboot i.e. call ESP.reset(). after you have the WiFi credentials saved. After a reboot you will be able to start a webserver again.

tzapu, 2016-05-08:
hi guys, you can also check the http server example, i am restarting a webserver there, after wifimanager is done, without needing a restart, hope it helps.

closed 2016-05-08 by tzapu

hawk88, 2020-05-27:
Hi,
I was able to setup and use your codes without any issues, except the http port is set to default 80.
However, due to my ISP is restricting ports 0-1000, and if I would to connect any device within my network from outside, I have to use NAT forwarding to a specific port, but I also have other devices using port 80 as well such as NAS storage. How to set http port other than port 80? Thanks.

alfo, 2020-05-27:
You don't need to change the port on the device. Just do port forwarding on your router:
Ext Port 8001 -> NAS Port 80
Ext Port 8002 -> ESP Port 80

tablatronix, 2020-05-27:
You can change ports in the alpha version