tzapu/WiFiManager #149: Station mode not necessary with previous saved credentials

piertoni, 2016-04-05:
Hi,
Imagine a router and ESP power down. The router maybe takes a while to span a wi-fi network.
In the meanwhile ESP boots and can't connect, then goes to station mode and doesn't connect anymore to router. This is not what I want for an IoT device.
Is possible to simply let ESP trying to connect without span station mode?

tzapu, 2016-04-06:
yes, either use `wifiManager.setConfigPortalTimeout(180);` so it stays in portal for said timeout, then make it reboot and retry (this is what i use) or just start config portal manually when you actually need it.

too many problems could arise otherwise

kentaylor, 2016-04-22:
Do not use the autoconnect functionality. Then it "Is possible to simply let ESP trying to connect to wi-fi without AP mode".

Start configuration mode manually when "you actually need it".

closed 2018-01-26 by tablatronix