tzapu/WiFiManager #150: Access wi-fi settings from station mode

piertoni, 2016-04-05:
Is it possible to access and change settings from station mode?

tzapu, 2016-04-06:
every time you call WiFi.begin(ssid,pass) it changes those settings

piertoni, 2016-04-06:
Sorry,
I meant if is it possible to access the settings webpage when the device is connected to router as station mode.

tzapu, 2016-04-06:
currently no, and there s a bug making it impossible to start it manually

kentaylor, 2016-05-02:
Using the forked version of this library at https://github.com/kentaylor/WiFiManager/ it is possible to access the settings page when the device is connected to a network via the network to which it is connected.

tzapu, 2016-05-08:
@piertoni it woulc be wonderful if you could test the fork that @kentaylor was so kind to provide and see if it works better for your needs.

reports are both welcome and needed :D
thanks

closed 2018-01-26 by tablatronix