tzapu/WiFiManager #151: AutoConnectWithFSParameters

kiralikbeyin, 2016-04-06:
Hi,

I am trying AutoConnectWithFSParameters example but i got these ;

nodemcu1.0 - 1.6.8 and stable esp 2.1

mounting FS...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

mounting FS...

Exception (9):
epc1=0x4022e03f epc2=0x00000000 epc3=0x00000000 excvaddr=0x401057a3 depc=0x00000000

ctx: sys 
sp: 3fff0160 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fff0300:  feefeffe feefeffe feefeffe feefeffe  
3fff0310:  feefeffe feefeffe feefeffe feefeffe  
3fff0320:  feefeffe feefeffe feefeffe feefeffe  
3fff0330:  40105784 feefeffe feefeffe feefeffe  
3fff0340:  4010579d 4010579a feefeffe feefeffe  
3fff0350:  400005e1 feefeffe feefeffe feefeffe  
3fff0360:  4022e03f 00000033 0000001a feefeffe  
3fff0370:  401057a3 40249f90 00000009 00000001  
3fff0380:  fbf8ffff 04000002 3feffe00 00000100  
3fff0390:  0000001a 00000018 04000102 40105784  
3fff03a0:  3fffc100 0000effe 00000009 00000009  
3fff03b0:  feefeffe feefeffe feefeffe feefeffe  
3fff03c0:  feefeffe feefeffe feefeffe feefeffe  
3fff03d0:  feefeffe feefeffe feefeffe feefeffe  
3fff03e0:  feefeffe feefeffe feefeffe feefeffe  
3fff03f0:  feefeffe feefeffe feefeffe feefeffe

tzapu, 2016-04-06:
try a SPIFFS.format before begin

make sure the FS include is the first one on top of everything else

make sure you set the proper SPIFFS size and memory size for your module

hope it helps

kiralikbeyin, 2016-04-06:
I tried this sketch and it didn't work.

Maybe a FS problem not related with your lib.

```
#include "FS.h"

void setup() {
  Serial.begin(115200);

  // always use this to "mount" the filesystem
  bool result = SPIFFS.begin();
  Serial.println("SPIFFS opened: " + result);

  // this opens the file "f.txt" in read-mode
  File f = SPIFFS.open("/f.txt", "w");
  
  if (!f) {
    Serial.println("File doesn't exist yet. Creating it");

    // open the file in write mode
    File f = SPIFFS.open("/f.txt", "w");
    if (!f) {
      Serial.println("file creation failed");
    }
    // now write two lines in key/value style with  end-of-line characters
    f.println("ssid=abc");
    f.println("password=123455secret");
  } else {
    // we could open the file
    while(f.available()) {
      //Lets read line by line from the file
      String line = f.readStringUntil('\n');
      Serial.println(line);
    }

  }
  f.close();
}

void loop() {
  // nothing to do for now, this is just a simple test

}
```

thoralt, 2016-04-14:
// this opens the file "f.txt" in read-mode
    File f = SPIFFS.open("/f.txt", "w");

You're not opening the file in read mode, but write mode instead. Use `"r"` as second parameter to `open()`.

closed 2018-01-26 by tablatronix