tzapu/WiFiManager #152: Checkbox to select DHCP or static

Jobenas, 2016-04-12:
Is there any way to include a checkbox in the manager so it is easy to select wether the station connection will be done with dhcp or static ip? I've been reading the documentation but haven't found an easy way of accomplishing this (actually haven't found any way of doing it). Thanks a lot!

Best regards,

Jorge.

Jobenas, 2016-05-05:
Hi Tzapu, 

I'm posting this on the same thread, hope you can reply :). I managed to place a checkbox on the config portal injecting html code as is shown in the documentation. My first question is how to get the value from the checkbox when I click on save, I tried using the getValue() method but can't get any actual value out of it. The second question is how to make the device connect with DHCP or static? because if I use setSTAStaticIPConfig that automatically forces a static IP to connect, but what if I need to give the choice to the user between the two? I've tried putting that last method in an if, but if the conditional is false, I don't get shown the parameters for IP and connects directly via DHCP. Thanks in advance for any hint you can provide me, and sorry if the post is a bit large.

Jorge.

tzapu, 2016-05-06:
jorge, there s no way yet to get a value if you use completly custom html

somewhere on the todo list...

kentaylor, 2016-05-08:
There is an example with a check box in the config portal at https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitchFS/ConfigOnSwitchFS.ino .

josep112, 2017-07-25:
Has anyone made any progress on this?

kgreene1983, 2018-02-05:
This a very useful thread. Ken... Will your code for checkbox work the tzapu version of wifi manager

Humancell, 2018-09-15:
I saw in the version 0.14 there is a commit that says "adds checkboxes", but I am unable to locate the documentation on this feature.  Where is this documented?  Just in issues ... and what specifically was added?

tablatronix, 2018-09-16:
adds checkboxes to the issue template not wm