tzapu/WiFiManager #156: Is it possible to access WiFi.softAPIP() in our sketch?

pieman64, 2016-04-22:
@tzapu I see WiFi.softAPIP() is passed with DEBUG_WM to the sketch but are we able to use the IP address as a variable / string in our sketch?

As you are aware there are occassions when ESP's don't give 192.168.4.1 as the IP address and we want to know when this occurs and without using Serial Monitor.

kentaylor, 2016-04-24:
DEBUG_WM is used to print to the serial port rather than pass data.

WiFi.softAPIP() can be called directly  from your sketch and the value returned can be used in your sketch but your sketch is blocked when the config portal is running.

pieman64, 2016-04-25:
As I thought @kentaylor so simple answer is we don't have access the actual IP allocated when working out in the field i.e. no Serial Monitor. Most of the time it allocates 192.168.4.1 as expected but not always. Not sure if it will show up on the routers GUI but we don't always have access to the router. Guess we can randomly ping possible IP's or perhaps tracert.

tzapu, 2016-04-25:
hi guys, 

you can use this https://github.com/tzapu/sive, #enter-config-mode to get the ip, print it on a lcd, or what not.

sorry for being unresponsive, it may be a while yet, but i hope to return to active duty on this asap
thank you

pieman64, 2016-04-25:
@tzapu that url is coming up with a 404 error.

tzapu, 2016-04-25:
hi, i corrected
https://github.com/tzapu/WiFiManager#enter-config-mode
cheers

pieman64, 2016-04-25:
Looks just the job< I should have looked closer at your guide. You are a start @tzapu.

tzapu, 2016-05-08:
thank you guys, doing some spring cleaning up so i will close this.
if any other issue, reopen please
thanks

closed 2016-05-08 by tzapu