tzapu/WiFiManager #158: Support Async TCP and Async Webserver

kaeferfreund, 2016-04-24:
Hi,

since I have switched to async tcp and web server by me-no-dev, things have gotten a lot faster. But sadly the WiFi Manager doesn't work anymore, bec its based on the "normal" WebServer Library.

Would appreciate if we could port the WiFi Manager to async

alanswx, 2016-07-02:
Here is a first version of this - it needs some work still:
https://github.com/alanswx/ESPAsyncWiFiManager
Also it should have a non-modal mode.

btomer, 2016-11-06:
If anyone is still interested in this, I wrote a WiFiManager fork that supports ESPAsyncWebServer and behaves exactly the same as the current version of WiFiManager (from master):
[](https://github.com/btomer/WiFiManager)

kentaylor, 2016-11-08:
I'm  interested, partly because I don't like [rebooting](https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino#L88) after WiFi manager completes . A WiFiManager that blocks until completion is a good design for some scenarios and I understand it will not with the ESPAsyncWebServer, is that correct? I see that you have a compiler directive to choose which web server version to use. Why was that design decision made rather than only having the ESPAsyncWebServer? Would you be willing to add your changes to [this version](https://github.com/tzapu/WiFiManager/) of WiFiManager? It is quite a bit different from the Tzapu version now and while I can figure out what you've done and copy, you would be able do it more easily.

tzapu, 2016-11-08:
@kaeferfreund adding async web server to wifimanager, while there s no async webserver in the core would mean people would need to add another library, which would raise the entry barrier. 
The way that @btomer did it though, could certainly sort that.

nice work @btomer , thanks. i m sure people will find it usefull. maybe i should compile all these links/branches in a list

@kentaylor i wrote a POC some time ago that lets you use a webserver with WiFiManager without reseting. I m sure it can be adapted to your sketches, see here https://gist.github.com/tzapu/ecc0759829d30d5a6152

closed 2018-01-26 by tablatronix

dronecz, 2020-07-24:
@tablatronix Is there any chance that async web server will be included in WiFiManager? I´m trying to change actual code to work with define as @btomer did, but so far no luck. Thx

tablatronix, 2020-07-24:
It should be easy to do in development alpha.m version

nferch, 2020-12-30:
rebased and updated @btomer 's patch against master branch a while back at https://github.com/nferch/WiFiManager/tree/asyncwebserver

tablatronix, 2020-12-30:
#794

tablatronix, 2020-12-30:
FYI this is in async branch based off development