tzapu/WiFiManager #159: Get current local IP address once connected to local network

syaffers, 2016-04-25:
Is there a way to get the local IP address of the ESP once it is connected to the local network? As of right now when I use wifiManager.autoConnect("ESP1") and enter my home network SSID and password it shows the "connecting page" and the mode changes and I can't ask the ESP what's my IP address unless I have a serial connection. 

I'm trying to use the ESP as a standalone device and when I push a button to call resetSettings() and connect to another wifi SSID, I shouldn't need to use the serial window to see my IP address but instead if I can see the IP address on the "connecting page" or on another page before the ESP changes to STA mode it would be better for my use case. Is there a way to do this natively using the library?

pieman64, 2016-04-25:
@syaffers yes see issue https://github.com/tzapu/WiFiManager/issues/156

tzapu, 2016-05-08:
thank you very much @pieman64  
i hope this is sorted @syaffers , if not please reopen the issue

thanks guys

closed 2016-05-08 by tzapu

syaffers, 2016-05-08:
Yes, sorted but forgot to close this. Thanks @pieman64 @tzapu

syaffers, 2016-05-13:
Hey @tzapu is there a way to show the IP address at the /wifisave page after successful connection? #156 only showed the APIP if connection is failed, what if connection is successful? Can I print the IP address on the /wifisave page?

kentaylor, 2016-05-13:
When connected this version  https://github.com/kentaylor/WiFiManager/ shows the IP number at the root page and the /info page. It also addresses a few other issues so I'm keen for people to try it and provide some feedback.

athleticmistud, 2018-12-24:
Is there a way that the kentaylor version can save the network name and password.  right now everytime the device is powered down you have to enter it all in again when it reboots.  I want it to save like the latest version of the wifimanager does as well as to show the ip address on  the save page