tzapu/WiFiManager #161: Once WiFiManager is used to create an AP, a new AP can't be created outside of WiFiManager.

rnearhoof, 2016-05-05:
Hello tzapu,  Thank you for creating your WiFiManager library, but I decided to give it a test run on a new adafruit huzzah board, and now I wish I hadn't. 

The library needs a clean way to remove all of the previously stored credentials. The function WiFiManager::resetSettings() is apparently supposed to do this, but it does not.

Here are the steps to reproduce the problem:
1.) From WiFiManager, call begin() in any fashion (with/without password or automatic SSID)

2.) After step 1, run a simple arduino ESP AP program that does not use WiFiManager. : You cannot create a new AP, only the previous AP created under WiFiManager.

3.) Reflash ESP with NodeMCU firmware (LUA) and try to create AP.  :You cannot create a new AP, but the previous AP created under WiFiManager is created.

4.) Erase 512K memory. : Old AP is still there.

5.) Try calling WiFiManager::resetSettings() : this does nothing to fix these issues.

Thanks for any help.

tzapu, 2016-05-05:
hi,

this is quite weird, maybe some flash corruption or something, as nothing like that should happen.

i can suggest 2 things, both in an empty sketch
1. user WiFi.begin("ssid", "pass"); to connect to a known ap, this should overwrite whatever settings you have
2. use
```
ESP.eraseConfig();
delay(1000);
ESP.reset();
delay(1000);
```
and then try whatever sketch you want

good luck

rnearhoof, 2016-05-05:
Thank you for the reply. I tried your suggestions, but no luck.  

It's not really an issue, since WiFiManager can overwrite the credentials, which makes me think it's not a flash corruption issue, though it still could always be.  I just have to use your library to set up the board I'll be running as a test bed.  

It is an interesting mystery though.  I have even tried finding and deleting config.json, but the phantom AP persists.  I read that the ESP can store 5 sets of credentials, but I thought they were for storing station mode login credentials, not AP setups.

It could also be my hardware. I have a few of the non-FCC cheap-o models that I can test with.
I will post any updates here.

closed 2018-01-26 by tablatronix