tzapu/WiFiManager #162: wifi save crashes again

tablatronix, 2016-05-05:
Ill have to narrow this down and follow up with esp8266 core ( using 2.2.0 atm )

```
Decoding 49 results
0x401003e9: _umm_malloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1380
0x40230770: etharp_output at ?? line ?
0x40231808: ip_output_if_opt at ?? line ?
0x401004f4: calloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1682
0x401073cc: pvPortZalloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 33
0x40220d4d: wifi_param_save_protect_with_check at ?? line ?
0x401057d2: spi_flash_read at ?? line ?
0x40220e35: system_param_save_with_protect at ?? line ?
0x40221224: wifi_station_ap_number_set at ?? line ?
0x4022137f: wifi_station_ap_number_set at ?? line ?
0x4022140c: wifi_station_ap_number_set at ?? line ?
0x40221448: wifi_station_set_config at ?? line ?
0x40202896: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x402027b7: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x4020a200: wifi_fpm_set_wakeup_cb at ?? line ?
0x40208d41: String::getBytes(unsigned char*, unsigned int, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402423d0: bignum_exptmod at ?? line ?
0x40208d41: String::getBytes(unsigned char*, unsigned int, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x40208d6c: String::indexOf(char, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402423d0: bignum_exptmod at ?? line ?
0x40208d90: String::indexOf(char) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402048b0: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40209638: Print::write(unsigned char const*, unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40208355: Print::write(char const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40241508: bignum_exptmod at ?? line ?
0x40208355: Print::write(char const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40220a9a: wifi_get_opmode at ?? line ?
0x402083a4: Print::println(__FlashStringHelper const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x4020276c: ESP8266WiFiSTAClass::status() at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x40206c4c: WiFiManager::connectWifi(String, String) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4020872c: String::changeBuffer(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x4020877b: String::reserve(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402087ad: String::copy(char const*, unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402088b4: String::operator=(String const&) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402010ae: delay at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_wiring.c line 53
0x4024153c: bignum_exptmod at ?? line ?
0x40207a4f: WiFiManager::startConfigPortal(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40241560: bignum_exptmod at ?? line ?
0x40207b85: WiFiManager::autoConnect(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40207bdc: WiFiManager::autoConnect() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x40206a5f: WiFiManager::WiFiManager() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x40202024: setup at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino line 51
0x40201f70: configModeCallback(WiFiManager*) at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino line 20
0x402091b4: loop_wrapper at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
```

__using pull request__
```
Decoding 30 results
0x40100255: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1298
0x4010020c: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x402441c0: bignum_exptmod at ?? line ?
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40105b32: spi_flash_read at ?? line ?
0x401075c0: vPortFree at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 18
0x40221a2a: wifi_param_save_protect_with_check at ?? line ?
0x402441c0: bignum_exptmod at ?? line ?
0x40221a47: wifi_param_save_protect_with_check at ?? line ?
0x40221aad: system_param_save_with_protect at ?? line ?
0x40221e9c: wifi_station_ap_number_set at ?? line ?
0x40221ff7: wifi_station_ap_number_set at ?? line ?
0x40222084: wifi_station_ap_number_set at ?? line ?
0x402220c0: wifi_station_set_config at ?? line ?
0x40202888: ESP8266WiFiSTAClass::disconnect(bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x4010053d: _umm_realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1489
:  (inlined by) realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1707
0x4010053d: _umm_realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1489
:  (inlined by) realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1707
0x40221712: wifi_get_opmode at ?? line ?
0x4022253b: wifi_station_get_connect_status at ?? line ?
0x40202990: ESP8266WiFiSTAClass::status() at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x40209338: String::changeBuffer(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402083bc: WiFiManager::startConfigPortal(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x4010020c: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40208528: WiFiManager::startConfigPortal() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x4020703f: WiFiManager::WiFiManager() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x40201fc4: setup at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\ConfigOnSwitch/ConfigOnSwitch.ino line 35
0x40209d90: loop_wrapper at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
```

kentaylor, 2016-05-05:
I'm not sure how to interpret this data but it looks like the device has crashed as instructed when ESP.reset(); has been called in code. The last line referenced is https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino#L51

The next line to run is ESP.reset(); at https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino#L54

Similarly in pull request #157
The last line referenced is https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino#L35
The next line to run is ESP.reset(); at https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino#L39

In the case of ConfigOnSwitch ESP.reset(); is called because the web server can only be started once per boot up. The ESP is restarted to make the web server functional when it is next started. It would be better if the web server could be started more than once per boot up but I'm not sure how to do that.

tzapu, 2016-05-06:
the one per boot problem is a bug i think in the cleanup of the webserver, it s worth reporting on the esp core team.

there is also another problem i ve notived with the change of ip on the various interfaces, sta/ap that the webserver doesn t like/behave

i ve managed to restart a webserver without a reboot after the wifimanager webserver goes down, maybe you can use this as inspiration
https://gist.github.com/tzapu/ecc0759829d30d5a6152

sorry, i am still otherwise engaged with daily stuff and can t give this the attention it deserves

tablatronix, 2016-05-06:
Yeah this happens when i save and it reboots i think and the. It crashes and in the pr version it gets stuck in loop crashing.
And seems to not connect first says 0 seconds spent.

tablatronix, 2016-05-07:
https://github.com/esp8266/Arduino/issues/1997

closed 2016-05-07 by tablatronix