tzapu/WiFiManager #165: Use EEPROM to store additional parameters ?

RudolfLapie, 2016-05-09:
I think WifiManager stores the SSID & Password by default in EEPROM. I would like to use the rest of the EEPROM for my own application. What part of the EEPROM is 'safe' to use or said differently, which bytes/offsets are used by WifiManager ?

kentaylor, 2016-05-09:
WiiManager does not store SSID and password in user space.

RudolfLapie, 2016-05-10:
Thanks Ken. That resolves it for me as it means, I can safely use all of the EEPROM for my device.

closed 2016-05-10 by RudolfLapie

pieman64, 2016-05-10:
Out of curiosity am I correct in thinking SPIFFS that WiFIManager uses for parameters (not SSID and PWD) is different to EEPROM?

kentaylor, 2016-05-10:
The version of WiFiManager at https://github.com/kentaylor/WiFiManager/ doesn't use SPIFFS. There is a memory map at https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map .

SPIFFS are in external SPI EEPROM as is the programs, data etc. At boot up part of the EEPROM is copied to RAM by the boot loader and then the user code is started.

pieman64, 2016-05-10:
@kentaylor looked at your notes for your fork.

I use @tzapu's original version quite a bit and none of my ESP's have ever bricked. I find the original to be very reliable.
Where does your version store additional parameters?

kentaylor, 2016-05-11:
@pieman64 When you say you have used the original "quite a bit" and found it to be "very reliable", I'd respond you must have been lucky. 

As of yesterday it was firmly established  that there is a fault in the underlying Espressif SDK that has caused WiFiManager to be unreliable. There were quite a few reports from other applications but it was WiFiManager failing and the ability of the Espressif developer @igrr to reproduce the failure that allowed the fault in the Espressif SDK fault to be identified. You will see that testing from @tablatronix shows both versions of WiFiManager failing but it is complicated. It is a developing situation and I'm not sure @igrr has sorted all of the issues yet. Follow the sequence of events at esp8266/Arduino#1997

Both versions store WiFi parameters in the same memory locations via Espressif SDK functions.

tzapu, 2016-05-11:
@kentaylor i don't think he was that lucky, for most people WiFiManager seems to be just fine. I myself have got around 6 modules running at all times, with power cuts, routers and internet going down that  seem to be just fine. a mix of battery and mains powered as well. Some have been running for almost a year, if not more.

At the same time, there have been users having connection issues, but judging by number of downloads, the number of people reporting issues, has been really really really small... Don t have other data to support this though, i can't know how many instances are in actual use.

I have had the umm_malloc bricking happen only once and had to erase the flash with esptool, and after sdk 2.1.0. Links helped me work through it then and added a patch to umm_malloc that seems to not sort it completly though. these issues are more likely to appear in development i think, when you keep switching settings, etc. "production" modules would be fine. I have to say i haven't done much development under sdk 2.2.0, so maybe the issues are a lot more prevalent now. 

also possible is that people did not update from SDK 2.0.0  and do not have these issues

tablatronix, 2016-05-11:
Yeah it always worked fine for me, the trouble with development is you do the same thing over and over with latest dev versions,

mihino, 2019-05-05:
Would be nice to have it directly in WiFiManager lib, but till then: check this post for EEPROM with WiFiManager: [https://homecircuits.eu/blog/esp8266-wifimanager-eeprom/](https://homecircuits.eu/blog/esp8266-wifimanager-eeprom/)

MasTerYoDa-60, 2023-02-28:
> WiiManager does not store SSID and password in user space.

So good!! Well, where this data is storage??? Thank you!!