tzapu/WiFiManager #166: Store multiple SSIDs ?

RudolfLapie, 2016-05-09:
I frequently move my esp8266 between different locations. Using WifiManager I now have to reconfigure the SSID & password each time. Would it be possible to store multiple (e.g. 3?) accesspoints that are tried one after the other before starting the accesspoint ?

kentaylor, 2016-05-10:
WiFiManager sets the ESP8266 SSID and password values using Espressif SDK functions. As far as I am aware there can not be multiple SSID's when done this way but it is convenient because the ESP8266 will join a WiFi network at anytime after power up that it becomes available and will rejoin if the WiFi network goes away and comes back later.

You could write code to connect to multiple SSIDs it but it is complicated unless you restrict trying different SSIDs to immediately after power up and that wouldn't be a good restriction.

RudolfLapie, 2016-05-10:
My first goal was simple and only during the "setup" phase of the device : store the last x successful SSID/password combinations and then at startup/power up,  check which of those appears in the list of SSIDs in that place at that moment. Then try to connect to the first one found.  If that doesn't work, start the AP and have the user configure the SSID/password for that place. So basically as it is today, but instead of 1 known SSID have x known SSIDs stored. A (hopefully?) small enhancement to the existing code. What you are suggesting would indeed be much much more powerful : when loosing connection, automatically reconnect to an available & known AP, but not what I suggested as it is indeed very complex to program.

kentaylor, 2016-05-10:
I looked into this a bit more. It turns out the device will store data for up to 5 WiFi access points. It will only use one of them rather than switch between them by itself but once one is selected it will handle the connectivity automatically. The Espressif SDK functions you would use are:- 
1. wifi_station_ap_number_set to set how many to save, might as well make it 5.
2. wifi_station_get_ap_info which returns details of the access points recorded and can be tested against the list of scanned access points.
3. wifi_station_ap_change to set which one to use.
See function definitions in sedctions 3.4.10 - 3.4.12 at http://www.mikrocontroller.net/attachment/245197/2C-SDK-Espressif_IoT_SDK_Programming_Guide_v0.9.5.pdf .

The algorithm when you save one would be choose the next empty one to save to. If there is already 5 saved, delete number 1. Then move 2 to 1 etc to make a vacancy at 5 and store the new one at 5.
At startup, scan for WiFi networks then change to the one from the scan list with the strongest signal that matches the SSID of one of the stored ones.

I like this idea but have no intention of implementing it. Perhaps someone else will be interested. Also it will have to wait until this bug https://github.com/esp8266/Arduino/issues/1997 discovered only a few hours ago is fixed in the Espressif SDK but that should be soon.

tablatronix, 2016-11-19:
Looks like this is not implemented in the arduino library wifimulti, it just auto selects best from a list you provide at runtime. You'd have to write your own implementation at this time.

kentaylor, 2016-11-19:
I wasn't aware of [WiFiMulti](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/station-examples.md) until referred to by @tablatronix who points out 

> it just auto selects best from a list.

So WiFiMulti doesn't use the flash SSID array but probably should.

When credentials are stored in flash the WiFi connection is maintained in the background . This leads to the advantages:-
- connection process starts before user code runs;
- connection is established if WiFi network becomes available after ESP device is started;
- connection is automatically reestablished if lost;
- all without user code.

WiFiMulti scans for an AP with the highest signal strength from a hard coded list and writes only the best one to flash. Thereafter this AP is used automatically until the ESP [fails to connect](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L45) at which point it repeats the process. 

If WiFiMulti was modified to use the SSID flash store rather than a hard coded list it could be used in WiFiManager to handle multiple SSIDs. I wonder if the ESP core team would be interested in such a modification?

tablatronix, 2016-11-19:
I did not even see an open issue so maybe we should create one, there was a closed issue but it was never seen by igrr and was not exactly feature request

implementing this is fairly complex, as you mentioned.

There is a list now, so you need add ap, remove ap, reorder ap? check already exists, connect to best ap, and all the UI necessary to perform this, it seems to be a large code complexity for such a minor feature.

tablatronix, 2016-11-19:
created feature request

Bergum, 2017-03-06:
I have done it this way to solve my problem. It works for me.

const char* ssid = "xxxxx";
const char* password = "xxxxx";
const char* ssid2 = "yyyyyy";
const char* password2 = "yyyyyy";

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
    
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  digitalWrite(LedYellow_1, HIGH);                // To give it indication of status
  Serial.println("LedYellow_1 On");      
    delay(30000);                                            // I give it 30 sec to connect, if it fails, try next

if (WiFi.status() != WL_CONNECTED) {
    WiFi.begin(ssid2, password2); 
    digitalWrite(LedYellow_2, HIGH);                // To give it indication of status
    digitalWrite(LedYellow_1, LOW);
    Serial.println("LedYellow_2 On");      
      delay(30000);                                               // I give it 30 sec to connect, if it fails.....
    }

else if (WiFi.status() != WL_CONNECTED) {
   Serial.print("Going DeepSleep");
    ESP.deepSleep(sleepTimeS * 1000000);         //  go deepsleep for 10 min and try all over again
 }

// I also terminate my script with deepsleep so that it will always start by scanning the network. 
// My use is to trigger the heater in my car at work or at home...

closed 2018-01-26 by tablatronix